dankeboy36/esp-exception-decoder #7: Arduino IDE 2.1.1

Azim, 2023-07-15:
The exception decoder terminal doesnt show up after running the command

dankeboy36, 2023-07-15:
Thank you for giving it a try and reporting back. It's expected. The extension API feature (https://github.com/arduino/arduino-ide/pull/2110) is available only from the nightly IDE builds: https://www.arduino.cc/en/software#nightly-builds

closed 2023-07-15 by dankeboy36

Azim, 2023-07-15:
I suggest adding a note about that in the installation section of readme to avoid similar issues.

on the side note, the command is still recognized by the IDE, just doesnt do anything. confusing.

dankeboy36, 2023-07-17:
OK. Let's investigate a bit. What's in the IDE logs? Please start the nightly Arduino IDE from a terminal with the `--log-level debug` flags, execute the command, nothing happens, quit IDE, and share the log files. You did not tell me any details about your env, but at the end of [this](https://github.com/arduino/arduino-ide/issues/1356#issuecomment-1482848858) comment, you will find where are the log files on all supported OSs. If it turns out to be a problem with the decoder extension, let's reopen this ticket. Thanks!