dankeboy36/esp-exception-decoder #8: No debug tab window appears on Arduino IDE 2.1.1

jponko, 2023-07-29:
No debug tab window appears on Arduino IDE 2.1.1
Using Linux Mint  21.2 o
exception decoder placed in here according to your directions while IDE was closed.
jponko@jponko-tc:~/.arduinoIDE/plugin$ ls
esp-exception-decoder-1.0.2.vsix

CTRL+SHIFT+p shows it in the list but no exception window appears when selected.
The IDE says this: Default VS Code API: 1.53.2
I don't use VS Code so I don't know if this is helpful.

System:
  Kernel: 5.15.0-78-generic x86_64 bits: 64 compiler: gcc v: 11.3.0 Desktop: Cinnamon 5.8.4
    tk: GTK 3.24.33 wm: muffin dm: LightDM Distro: Linux Mint 21.2 Victoria base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: LENOVO product: 10FLS02D00 v: ThinkCentre M900
    serial: <superuser required> Chassis: type: 3 serial: <superuser required>
  Mobo: LENOVO model: 30D0 v: SDK0J40697 WIN 3305025634376 serial: <superuser required>
    UEFI: LENOVO v: FWKTB9A date: 08/17/2021
CPU:
  Info: quad core model: Intel Core i5-6600T bits: 64 type: MCP arch: Skylake-S rev: 3 cache:
    L1: 256 KiB L2: 1024 KiB L3: 6 MiB
  Speed (MHz): avg: 800 min/max: 800/3500 cores: 1: 800 2: 800 3: 800 4: 800 bogomips: 21599

dankeboy36, 2023-07-30:
> on Arduino IDE 2.1.1

Thank you for the bug report. Please try it with the nightly Arduino IDE 2.x build and let me know how it works: https://www.arduino.cc/en/software#nightly-builds.

jponko, 2023-07-31:
Yes, the exception decoder window now works with the IDE current nightly
build but it is unable to decode the register dump. (see attached
screenshot)

dankeboy36, 2023-07-31:
> Yes, the exception decoder window now works with the IDE current nightly

Great. The decoder does not work with 2.1.1, but the nightly Arduino IDE 2.x build.

> build but it is unable to decode the register dump. (

What's the error? Can you share your environment details: the sketch, the selected board, and maybe the error message? 

> (see attached screenshot)

You did not attach anything. Please do not attach a screenshot, but copy the error message from the decoder console. Thanks!

jponko, 2023-07-31:
Actually it does work, I made the mistake of not including the backtrace
line when I did the copy and paste. Here's the output:

ESP Exception Decoder
Sketch: NJAD_VFO_V1.1 FQBN: esp32:esp32:nodemcu-32s

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was
unhandled.

Core  1 register dump:
PC      : 0x400d1c15  PS      : 0x00060a30  A0      : 0x800da5dd  A1      :
0x3ffb27e0
A2      : 0x3ffc2750  A3      : 0x00000000  A4      : 0x00000080  A5      :
0x3ffc14a0
A6      : 0x3ffc14a4  A7      : 0x3ffc14a8  A8      : 0x00000000  A9      :
0x3ffb27c0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     :
0x00000003
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE:
0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40086084  LEND    : 0x4008608f  LCOUNT  :
0xffffffff

Backtrace:0x400d1c12:0x3ffb27e00x400da5da:0x3ffb2820

PC: 0x400d1c15: setup() at
/media/jponko/odyssey/AC9NM/It-s-Not-Just-Another-Digital-VFO-by-WA2FZW-VK3PE-and-G3ZQC-Version-1.1/NJAD_VFO_V1.1/NJAD_VFO_V1.1.ino:359
EXCVADDR: 0x00000000

Decoding stack results
0x400d1c12: setup() at
/media/jponko/odyssey/AC9NM/It-s-Not-Just-Another-Digital-VFO-by-WA2FZW-VK3PE-and-G3ZQC-Version-1.1/NJAD_VFO_V1.1/NJAD_VFO_V1.1.ino:359
0x400da5da: loopTask(void*) at
/home/jponko/.arduino15/packages/esp32/hardware/esp32/2.0.5/cores/esp32/main.cpp:42

Paste exception to decode...

----------------------------------------------------------------------------------------------------
Thanks for fixing the exception decoder. I don't use it for my own code but
it helps with diagnosiing others code.

dankeboy36, 2023-07-31:
> Actually it does work,

I am glad it's working now. I am going to close this issue. If you do not agree, please let me know.

----

> on Arduino IDE 2.1.1

> > Yes, the exception decoder window now works with the IDE current nightly
> 
> Great. The decoder does not work with 2.1.1, but the nightly Arduino IDE 2.x build.

See related: https://github.com/dankeboy36/esp-exception-decoder/issues/7#issuecomment-1636816401.

closed 2023-07-31 by dankeboy36