dankeboy36/esp-exception-decoder #10: No board selected

majodi, 2023-09-19:
I get "No board selected" in red. 
[image]

birdoffire1549, 2023-09-20:
I too have this same issue.
Using Arduino IDE:

Version: 2.2.1
Date: 2023-08-31T14:26:39.874Z (2 weeks ago)
CLI Version: 0.34.0

dankeboy36, 2023-09-20:
- Open the _Command Palette_ with <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>.
 - Type `Preferences: Open Settings (UI)` and press <kbd>Enter</kbd>.
 - Type `arduinoAPI.log` into the _Search Settings_ `<input>`.
 - Tick the checkbox next to the preference value.
   [image]
 - Open the _Command Palette_ and type `Show Output Channel...` and press <kbd>Enter</kbd>.
 - You will see (at least) two channels: `Arduino API` and `ESP Exception Decoder (Log)`. If you select one of them, the default _Output_ channel in the Arduino IDE changes, and you will see additional logs.

https://github.com/dankeboy36/esp-exception-decoder/assets/111981763/88a4a8b1-1600-4cb7-839a-aefd1030d025
 - Select the `Arduino API` output channel and click the `Clear Output` toolbar in the _Output_ view to clear the content. Do the same in the `ESP Exception Decoder (Log)` channel.
   [image]
 - Unset the board+port selection in the Arduino IDE: open the _Command Palette_ and execute the `Developer (Arduino): Clear the Board and Port Selection` command. The Arduino IDE will forget your board and port selection in this sketch.
 - (Re)Select the board and port the usual way you do in the Arduino IDE.
 - Open the `Arduino API` output channel, copy the log output, and attach it to this issue.
 - Open the `ESP Exception Decoder (Log)` output channel, and share the log messages with me.
 - Open the _Command Palette_ and execute the `Developer (Arduino): Dump the Board List` command. The IDE will print a JSON to the _Output_ view and share that output with me.

Thank you!

birdoffire1549, 2023-09-20:
@dankeboy36, The unsetting the board and port option and then re-selecting it has fixed the issue for me.

dankeboy36, 2023-09-21:
If somebody can reproduce it, please provide the logs. I will look into the defect. I will keep this issue open.

per1234, 2023-09-21:
The fault occurs when the board was selected on IDE startup:

### To reproduce

1. Select **File > New Sketch** from the Arduino IDE menus.
1. Select a board from the Arduino IDE menus.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch to any convenient name and location.
   **ⓘ** This step is performed in order to associate the board selection with a sketch that can be restored on IDE startup.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start Arduino IDE.
   The saved sketches that were open during the previous session will be restored.
1. Select the window of the sketch you saved during the previous steps.
1. Run the "**ESP Exception Decoder: Show Decoder Terminal**" command.
   🐛 The Exception Decoder view shows the message "`No board selected`" even though a board is in fact selected.
1. Select any other board from the Arduino IDE menus.
1. Select the target board once more.
   🙂 The Exception Decoder now recognizes the board selection.

I am happy to provide logs, but I think you will be able to easily reproduce the fault now from the instructions I provided @dankeboy36 so they probably won't be needed.

---

I notice the fault occurs under the same conditions as https://github.com/arduino/arduino-ide/issues/2230#issuecomment-1728823010

majodi, 2023-09-21:
Selecting another board and then the right board gave me "Sketch path is not set". Then after opening another sketch, close it and re-open the right sketch finally the exception decoder worked. So it seems that when the IDE is started with an existing project (sketch and board) it will not communicate the board/sketch to your code. I don't know (and don't have time to dive into it) if there is a way to force the IDE API to refresh and give you the current active board/sketch but that's what you need I guess.

kittaakos, 2023-09-21:
This must be fixed in the IDE2 code. The problem is the same as https://github.com/arduino/arduino-ide/pull/2165#pullrequestreview-1568872036 and https://github.com/arduino/arduino-ide/pull/2165#pullrequestreview-1571165467. The Arduino state is set by IDE2 too early, and the board+port settings still need to be restored. [Here](https://github.com/arduino/arduino-ide/pull/2165/files#diff-e95a72cd7e218490c80be1426a73803c2928e0c6e6bc1097177a5d526c732499L78-R79) is the overlooked `TODO` in the PR 🤦  This should fix:

```diff
diff --git a/arduino-ide-extension/src/browser/contributions/update-arduino-state.ts b/arduino-ide-extension/src/browser/contributions/update-arduino-state.ts
index 35b4c2ab..4e575430 100644
--- a/arduino-ide-extension/src/browser/contributions/update-arduino-state.ts
+++ b/arduino-ide-extension/src/browser/contributions/update-arduino-state.ts
@@ -76,7 +76,9 @@ export class UpdateArduinoState extends SketchContribution {
   }
 
   override onReady(): void {
-    this.updateBoardsConfig(this.boardsServiceProvider.boardsConfig); // TODO: verify!
+    this.boardsServiceProvider.ready.then(() => {
+      this.updateBoardsConfig(this.boardsServiceProvider.boardsConfig); // TODO: verify!
+    });
     this.updateSketchPath(this.sketchServiceClient.tryGetCurrentSketch());
     this.updateUserDirPath(this.configService.tryGetSketchDirUri());
     this.updateDataDirPath(this.configService.tryGetDataDirUri());
```

dankeboy36, 2023-09-27:
Please use the Arduino IDE 2.x nightly build with the fix: https://www.arduino.cc/en/software#nightly-builds

closed 2024-02-08 by dankeboy36