dankeboy36/esp-exception-decoder #11: Arduino IDE 2.2.1 Board: NodeMCU ESP8266 Lolin v3(ESP-12E). Stacktrace isn't recognized

r-hmn, 2023-10-24:
Hi dankeboy36

i'm posting here as you advised [here](https://forum.arduino.cc/t/exception-decoder/1055759/7?u=ardihmn)
in that link it shows the settings and board i use.
(i also tested by enabling the "C++ exceptions")
Aside from the mentioned nightly build, i use Arduino IDE 2.2.1. Both provide the same result.

In case it's interesting my build output is this (i replaced ascii-art with '|---' etc)
```
. Variables and constants in RAM (global, static), used 39336 / 80192 bytes (49%)
|-   SEGMENT  BYTES    DESCRIPTION
|------- DATA     1516     initialized variables
|------- RODATA   5164     constants       
|------- BSS      32656    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 60755 / 65536 bytes (92%)
|-   SEGMENT  BYTES    DESCRIPTION
|------- ICACHE   32768    reserved space for flash instruction cache
|------- IRAM     27987    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 322768 / 1048576 bytes (30%)
|-   SEGMENT  BYTES    DESCRIPTION
|------- IROM     322768   code in flash   
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:3f:29:4f
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 361584 bytes to 261258...
Writing at 0x00000000... (6 %)
Writing at 0x00004000... (12 %)
Writing at 0x00008000... (18 %)
Writing at 0x0000c000... (25 %)
Writing at 0x00010000... (31 %)
Writing at 0x00014000... (37 %)
Writing at 0x00018000... (43 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (56 %)
Writing at 0x00024000... (62 %)
Writing at 0x00028000... (68 %)
Writing at 0x0002c000... (75 %)
Writing at 0x00030000... (81 %)
Writing at 0x00034000... (87 %)
Writing at 0x00038000... (93 %)
Writing at 0x0003c000... (100 %)
Wrote 361584 bytes (261258 compressed) at 0x00000000 in 23.0 seconds (effective 125.8 kbit/s)...
Hash of data verified.

Leaving...
```
The exeption happens after the 2nd time the ESP8266WebServer receives a POST request and writes a file to LittleFS.
My tracktrace is this (actually two?.. happening directy after eachother)
```

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:191 __yield

>>>stack>>>

ctx: sys
sp: 3ffff570 end: 3fffffb0 offset: 0010
3ffff580:  00000000 00000001 00000000 40211905  
3ffff590:  00000000 3fff05c8 3fff104c 40206b13  
[... 198 lines cut ...]
3ffff6e0:  000017a0 000002f4 00000000 000f000f  
3ffff6f0:  00000000 00000000 000f0000 00000000  
3ffff700:  00000000 000f0000 00000000 6bdae720  
3ffff710:  00000000 00000000 3fff0ec8 3fff1248  
3ffff720:  3fffdad0 3fff0efc 3fff0ec8 4020a82a  
3ffff730:  40214698 00000000 00001388 4020c47a  
3ffff740:  00000000 00000000 3fff31cc 00000000  
3ffff750:  3fffdad0 3fff1080 3fff121c 3fff1248  
3ffff760:  3fffdad0 00000000 3fff121c 4020a967  
3ffff770:  40209ac8 feefeffe 00000001 3fff0eb8  
3ffff780:  40209ac8 00000000 40213b38 40100114  
3ffff790:  00000001 00000000 00000000 00000000  
3ffff7a0:  3fc00000 40000000 0000005f 3fff0da4  
3ffff7b0:  00010001 42be0000 00000048 feefeffe  
3ffff7c0:  feefeffe 00000000 74207755 74736b65  
3ffff7d0:  36302320 00000000 00000000 00000000  
3ffff7e0:  00000000 00000000 00000000 00000000 
3ffff7f0:  00000000 00000000 00000000 00000000  
3ffff800:  00000000 00000000 00000000 00000000  
3ffff810:  00000000 00000000 00000000 00000000 

Could not recognize stack trace/backtrace

Paste exception to decode...
```

dankeboy36, 2023-10-24:
I looked into it, @r-hmn, and I could reproduce the same behavior, but I do not think this is a decoder bug. Your stack dump looks invalid. See [here](https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html), ESP8266 stack dump starts with `>>>stack>>>` and ends with `<<<stack<<<`. Yours not.

Your starts with this:

```

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:191 __yield

>>>stack>>>

ctx: sys
sp: 3ffff570 end: 3fffffb0 offset: 0010
3ffff580:  00000000 00000001 00000000 40211905  
```

And another stack dump comes in before the other dump is fully printed(?):
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:191 __yield

>>>stack>>>

ctx: sys
sp: 3ffff290 end: 3fffffb0 offset: 0010
3ffff2a0:  00000000 00000000 00000000 40103132  
```

You should ask the Arduino forum (https://forum.arduino.cc/c/software/ide-2-x/93) why you see two stack dumps. Share the board and your sketch (if you can) with the community. The decoder will work once you figure out how to have one stack dump.

dankeboy36, 2023-10-24:
> another stack dump comes in before the other dump is fully printed(?):

Maybe it's due to the serial monitor in IDE2: https://github.com/arduino/arduino-ide/issues/1081.

You can print the monitor to an actual terminal and copy it back to the decoder terminal.

 - open IDE2,
 - make sure the monitor widget is not open, and the monitor is not running,
 - select the board, verify, and upload from IDE2 as usual,
 - remember the port your board is attached to (mine is `/dev/cu.usbmodem14101` for this example),
[image]
 - ([from the screenshots](https://forum.arduino.cc/t/exception-decoder/1055759/7), I am assuming you're on Linux) navigate to the `resources/app/lib/backend/resources` folder inside your IDE2 installation,
 - you see the `arduino-cli` executable there. if you run `./arduino-cli version`, it prints the version
 - if you run `./arduino-cli monitor -p /dev/cu.usbmodem14101` (your port is probably different), you will see the monitor output in the terminal. Here is my example (I am on macOS, though):

```
% pwd
/Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/resources
% ./arduino-cli version
arduino-cli  Version: 0.34.0 Commit: 304d48cd Date: 2023-08-22T15:54:20Z
% ./arduino-cli monitor -p /dev/cu.usbmodem14101
Connected to /dev/cu.usbmodem14101! Press CTRL-C to exit.
count=10000000, lines/sec=0
count=10000001, lines/sec=0
count=10000002, lines/sec=0
count=10000003, lines/sec=0
count=10000004, lines/sec=0

```
Now, you can check if you really see two stack dumps overwriting each other. If you see the dump between `>>>stack>>>` and `<<<stack<<<`. It's an IDE2 bug with the monitor, and the decoder should work. If you see two starting `>>>stack>>>` tags but no trailing one, I would ask on the Arduino forum.

I hope this helps.

r-hmn, 2023-10-24:
The github you listed is about the selection itself changing. it's quite an anoyance indeed. Thanks for the aduino-cli explanation. that will come in handy the next time. 

The stacktrace which i mentioned is how it is listed on the output. Some times there's one stacktrace, sometime i get two. it's probably an indicator of how bad the program is failing :)

i copied it in multiple parts to get the whole thing properly reported here. But when i post it to the exception decoder, i use one stacktrace at the time, starting after the `---CUT..`

r-hmn, 2023-10-24:
> I looked into it, @r-hmn, and I could reproduce the same behavior, but I do not think this is a decoder bug. Your stack dump looks invalid. See [...]

> You should ask the Arduino forum (https://forum.arduino.cc/c/software/ide-2-x/93) why you see two stack dumps. Share the board and your sketch (if you can) with the community. The decoder will work once you figure out how to have one stack dump

Thanks. i will.
Cheers

r-hmn, 2023-10-29:
Update; I posted my problem [here](https://forum.arduino.cc/t/my-esp8266-stacktrace-is-malformed-what-am-i-doing-wrong/1182664/6). 
Turned out i was writing outside a buffer and my stack sometimes overwritten because memory/stack is quite full, which got my stack corrupted, hence strange errors.

r-hmn, 2023-10-29:
Update; I posted my problem [here](https://forum.arduino.cc/t/my-esp8266-stacktrace-is-malformed-what-am-i-doing-wrong/1182664/6). 
Turned out i was writing outside a buffer and my stack sometimes got overwritten because memory/stack is quite full, hence strange errors happened.

closed 2023-10-29 by r-hmn

r-hmn, 2023-11-03:
Would just appending the text `<<<stack<<<`  be a way to let the decoder recognize and decode the top part of the stack?

dankeboy36, 2023-11-04:
> Would just appending the text `<<<stack<<<` be a way to let the decoder recognize

Yes. I did the same with your initial example (https://github.com/dankeboy36/esp-exception-decoder/issues/11#issue-1959203954). If added the closing `<<<stack<<<`, the decoder could recognize and parse the stack.

> and decode the top part of the stack?

I think, yes. I am unsure whether it's correct or complete, though.

tornado67, 2025-07-06:
Just in case you need it I created a python script to deocde the exception stack https://github.com/tornado67/yaeed :)