dankeboy36/esp-exception-decoder #12: Could not detect any instruction adress in the stack/backtrace

db1992ger, 2023-11-21:
First of all, thanks for your plugin!

I tried to decode the register dump I get, but the decoder gives an error message:

"Could not detect any instruction addresses in the stack trace/backtrace"

[image]

Is this a bug or am I missing something?

Thanks in advance!

dankeboy36, 2023-11-21:
Thanks for reporting it. Please attach the stack trace you pasted into the decoder terminal to the issue, preferably as a [code block comment](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#fenced-code-blocks). Otherwise, it will be hard to reproduce the defect from a screenshot. I will take a look at it.

db1992ger, 2023-11-21:
Thanks for your fast response! 

```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060a30  A0      : 0x00000000  A1      : 0x3ffd1a90  
A2      : 0x3ffd5c40  A3      : 0x3ffbdbac  A4      : 0x00000005  A5      : 0x00000021  
A6      : 0x3ffc60d4  A7      : 0x00000000  A8      : 0x800d2649  A9      : 0x3ffd1a70  
A10     : 0x3ffd3928  A11     : 0x00000000  A12     : 0x3ffbdbaf  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffd1a40  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400d26cc  LEND    : 0x400d271a  LCOUNT  : 0x00000004  

Backtrace: 0xfffffffd:0x3ffd1a90

```
Let me know if there's anything else that helps you helping me^^

dankeboy36, 2023-11-21:
I debugged into the code. From your decoder input, the decoder extracts this string to do further match for instruction addresses:

```
 0xfffffffd:0x3ffd1a90
```

but it fails to find any because they do not match the expected instruction address regex: `/4[0-3][0-9a-f]{6}\b/gim` ([Here](https://github.com/me-no-dev/EspExceptionDecoder/blob/ff4fc36bdaf0bfd6e750086ac01554867ede76d3/src/EspExceptionDecoder.java#L483-L484) is the logic from the original implementation and the same [here](https://github.com/dankeboy36/esp-exception-decoder/blob/c89a25959ed50fcbae026a07cd8c02b77e51b445/src/decoder.ts#L371) from this repo.)

This is where my domain knowledge is limited; I can easily relax the regex to match the `0xfffffffd` (from your use case), but I do not know if this is correct.

If somebody can chime in and help with why there are no instruction addresses in the stack trace, I would be happy to fix it if this is a bug.

db1992ger, 2023-11-21:
> This is where my domain knowledge is limited; I can easily relax the regex to match the 0xfffffffd (from your use case), but I do not know if this is correct.

We could try that with my sketch and see if the result makes sense. I have somewhat of an idea of where the exeption is coming from.

dankeboy36, 2023-11-22:
Can you share your sketch (or its simplified version) that produces the problematic backtrace? I am open to providing you with a tester VSIX with the changes, but I would like to understand the problem better.

If I blindly relax the regex, people might suddenly see trash in the stack trace. Here is another (bash) tool to decode ESP32 backtrace. It also [matches only the registered address](https://github.com/tve/esp32-backtrace/blob/eaab9bf0fe89aa3c868e68d991d576767121e35b/esp32-backtrace#L66-L76).

db1992ger, 2023-11-22:
Sure, I will probably have some spare time to find the exact error tomorrow, so that I can give you the simplified version of it. I bet it's some kind of data corruption by writing beyond the bounderies of an array with pointers in a for loop.  I'll keep you updated!

db1992ger, 2023-11-24:
I think I found the exact place where the programm crashes. Due to the fact that the sketch is quite large, I''ll simplify it as much as I can:

Imagine a class:
```
class aclass{
       bool set;
       void action(){
           Serial.println("test");
          // other stuff;
       };
}

aclass aclass_array[100];

loop{

  for(int i = 0; i < 100; i++){
       if(aclass_array[i].set){
               Serial.print("i = ");
               Serial.println(i); //When it chrashes, this is the last working line, although "i" is "valid"
               aclass_array[i].action();
       }
  
   }
 //other stuff
}
```
The strange thing is, that everthing is executed and working fine for quite some time but then suddenly it crashes. I put some "serial.prints" here and there and found out that even the Serial.print at the beginning of the function isn't executed anymore, which means that after some time "aclass_array[i].action();" becomes an "IllegalIntructionFetch". Im still working on where this problem comes from exactly. As I said before, I think other stuff in the programm accidently overwrites memory.  Maybe that explains why the Decoder can't find instruction adresses in the backtrace.

closed 2023-11-25 by db1992ger