dankeboy36/esp-exception-decoder #14: Segmentation fault when decoding exeption for ESP8266 generic module

ffpp2003, 2024-01-26:
After pasting this exeption:
```
Exception (3):
epc1=0x402203e6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4026c7d1 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe30 end: 3fffffd0 offset: 0150
3fffff80:  3ffe8bb8 3fff05fc 3fff03e8 40208a38  
3fffff90:  3ffe8baa 3ffe9150 3fff05fc 3fff0788  
3fffffa0:  3fffdad0 0000000f 3fff05fc 3fff0788  
3fffffb0:  3fffdad0 00000000 3fff075c 4020cca8  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100e7d  
<<<stack<<<
```

I got a segmentation fault on the arduino terminal:
```
Command was killed with SIGSEGV (Segmentation fault): /home/fpp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gdb --batch /tmp/arduino/sketches/3EE82EAA269B26DE793A1A57DCC72733/AlarmAddonv2.ino.elf -ex set listsize 1 -ex list *0x40208a38 -ex list *0x4020cca8 -ex list *0x40100e7d -ex q
```

This is what I can see on the terminal when launching arduinoIDE from it:
```
2024-01-26T15:27:42.930Z root INFO [hosted-plugin: 1714] 0x402203e6 is in strnlen (/workdir/repo/newlib/newlib/libc/string/strnlen.c:38).
2024-01-26T15:27:42.933Z root ERROR [hosted-plugin: 1714] 38	/workdir/repo/newlib/newlib/libc/string/strnlen.c: No existe el fichero o el directorio.
2024-01-26T15:27:42.962Z root ERROR [hosted-plugin: 1714] Failed to execute: /home/fpp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gdb ["--batch","/tmp/arduino/sketches/3EE82EAA269B26DE793A1A57DCC72733/AlarmAddonv2.ino.elf","-ex","set listsize 1","-ex","list *0x40208a38","-ex","list *0x4020cca8","-ex","list *0x40100e7d","-ex","q"] Error: Command was killed with SIGSEGV (Segmentation fault): /home/fpp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gdb --batch /tmp/arduino/sketches/3EE82EAA269B26DE793A1A57DCC72733/AlarmAddonv2.ino.elf -ex set listsize 1 -ex list *0x40208a38 -ex list *0x4020cca8 -ex list *0x40100e7d -ex q
    at P (/tmp/vscode-unpacked/esp-exception-decoder-1.0.2.vsix/extension/dist/extension.js:1:41810)
    at /tmp/vscode-unpacked/esp-exception-decoder-1.0.2.vsix/extension/dist/extension.js:1:49104
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async t.run (/tmp/vscode-unpacked/esp-exception-decoder-1.0.2.vsix/extension/dist/extension.js:1:31779)
    at async C (/tmp/vscode-unpacked/esp-exception-decoder-1.0.2.vsix/extension/dist/extension.js:1:23123)
    at async Promise.all (index 2)
    at async t.decode [as decoder] (/tmp/vscode-unpacked/esp-exception-decoder-1.0.2.vsix/extension/dist/extension.js:1:20034)
    at async m.decode (/tmp/vscode-unpacked/esp-exception-decoder-1.0.2.vsix/extension/dist/extension.js:1:28579) {
  shortMessage: 'Command was killed with SIGSEGV (Segmentation fault): /home/fpp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gdb --batch /tmp/arduino/sketches/3EE82EAA269B26DE793A1A57DCC72733/AlarmAddonv2.ino.elf -ex set listsize 1 -ex list *0x40208a38 -ex list *0x4020cca8 -ex list *0x40100e7d -ex q',
  command: '/home/fpp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gdb --batch /tmp/arduino/sketches/3EE82EAA269B26DE793A1A57DCC72733/AlarmAddonv2.ino.elf -ex set listsize 1 -ex list *0x40208a38 -ex list *0x4020cca8 -ex list *0x40100e7d -ex q',
  escapedCommand: '"/home/fpp/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gdb" --batch "/tmp/arduino/sketches/3EE82EAA269B26DE793A1A57DCC72733/AlarmAddonv2.ino.elf" -ex "set listsize 1" -ex "list *0x40208a38" -ex "list *0x4020cca8" -ex "list *0x40100e7d" -ex q',
  exitCode: undefined,
  signal: 'SIGSEGV',
  signalDescription: 'Segmentation fault',
  stdout: '',
  stderr: '',
  failed: true,
  timedOut: false,
  isCanceled: false,
  killed: false
}
```

About arduinoIDE:
```
Version: 2.2.2-nightly-20240125
Date: 2024-01-25T03:08:37.815Z
CLI Version: 0.35.1

Copyright © 2024 Arduino SA
```

dankeboy36, 2024-01-29:
Thank you for the bug report. It seems to me a problem with `xtensa-lx106-elf-gdb` and not the decoder extension. Probably, it cannot be fixed here, but if you're willing to participate, we can try to figure out what could be the problem or whether it could be resolved with a newer version of gdb. The current version is almost five years old.

```
~/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-gdb --version
GNU gdb (GDB) 8.2.50.20190202-git
```

A more recent version could contain the bug fix for this issue. We could also try to reproduce it with the [1.8.x Arduino IDE](https://www.arduino.cc/en/software#legacy-ide-18x) using the [original implementation](https://github.com/me-no-dev/EspExceptionDecoder) of this extension.

ffpp2003, 2024-01-31:
I was able to fix the exeption, and I dont have a copy of the faulty code sadly.
If I get any other exeption and a segfault when trying to decode it, I will post the code here so I can take a look at what you suggest to try to find out whats causing the segfault.

dankeboy36, 2024-01-31:
Thank you so much for coming back and giving an update. For now, I will close this issue. Please let me know if the error pops up again, and we reopen it.

closed 2024-01-31 by dankeboy36