dankeboy36/esp-exception-decoder #15: Could not detect the DGB tool path

chked, 2024-02-20:
Could not detect the DGB tool path message appears in Exception Decoder Output window. IDE is 2.3.1, board ESP8266.
How debug output can be enabled? This could be helpful to find out such issures.

dankeboy36, 2024-02-20:
Thank you for reporting it.

> Could not detect the DGB tool path message appears in Exception Decoder Output window

Please send the complete decoder output as a [markdown string](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code).

> IDE is 2.3.1, board ESP8266.

It would be interesting to know what your board selection IDE2 is. You can make a screenshot, but executing the `(Developer) Arduino: Dump the Board List ` command and pasting the output here is even better.

To execute a command:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**(Developer) Arduino: Dump the Board List**" command from the menu.
1. Press <kbd>**Enter**</kbd>.

Further questions:
 - Which platform do you use, and which version?
 - Does verify/upload work in the Arduino IDE?
 - Do you have manually installed cores (in your sketchbook)?

chked, 2024-02-20:
The decoder output is 
1. After start
_ESP Exception Decoder
Sketch: Test FQBN: esp8266:esp8266:esp12
The summary of the previous compilation is unavailable. Compile the sketch_
2. After compiling the sketch
_ESP Exception Decoder
Sketch: Test FQBN: esp8266:esp8266:esp12
Could not detect the DGB tool path_

Used versions:
Arduino IDE 2.3.1
esp8266 community 3.1.2

Verify/Upload works.

There are no manually installed cores.

kittaakos, 2024-02-20:
> 1. FQBN: esp8266:esp8266:esp12

Where does this FQBN come from?

```sh
cat ~/.arduinoIDE/arduino-cli.yaml
```

```yaml
board_manager:
    additional_urls:
        - https://arduino.esp8266.com/stable/package_esp8266com_index.json
        - https://descartes.net/package_drazzy.com_index.json
        - https://downloads.arduino.cc/packages/package_nano_esp32_index.json
        - https://downloads.arduino.cc/packages/package_silabs_beta_index.json
        - https://espressif.github.io/arduino-esp32/package_esp32_index.json
        - https://github.com/e-radionicacom/Croduino-Board-Definitions-for-Arduino-IDE/raw/master/package_Croduino_Boards_index.json
        - https://github.com/e-radionicacom/Dasduino-Board-Definitions-for-Arduino-IDE/raw/master/package_Dasduino_Boards_index.json
        - https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json
build_cache:
    compilations_before_purge: 10
    ttl: 720h0m0s
daemon:
    port: "50051"
directories:
    builtin:
        libraries: /Users/a.kitta/Library/Arduino15/libraries
    data: /Users/a.kitta/Library/Arduino15
    downloads: /Users/a.kitta/Library/Arduino15/staging
    user: /Users/a.kitta/Documents/Arduino
library:
    enable_unsafe_install: false
locale: en
logging:
    file: ""
    format: text
    level: info
metrics:
    addr: :9090
    enabled: true
output:
    no_color: false
sketch:
    always_export_binaries: false
updater:
    enable_notification: true
```

```sh
~/dev/git/arduino-ide/arduino-ide-extension/src/node/resources/arduino-cli version
```

```sh
arduino-cli  Version: 0.36.0-rc.1 Commit: 31e03410 Date: 2024-01-11T12:36:53Z
```

```sh
~/dev/git/arduino-ide/arduino-ide-extension/src/node/resources/arduino-cli core list --format json --config-file ~/.arduinoIDE/arduino-cli.yaml | jq '.platforms[] | select(.id == "esp8266:esp8266").releases[] | select(.version == "3.1.2").boards'
```

```json
[
  {
    "name": "ThaiEasyElec's ESPino",
    "fqbn": "esp8266:esp8266:espinotee"
  },
  {
    "name": "Amperka WiFi Slot",
    "fqbn": "esp8266:esp8266:wifi_slot"
  },
  {
    "name": "ESPresso Lite 1.0",
    "fqbn": "esp8266:esp8266:espresso_lite_v1"
  },
  {
    "name": "XinaBox CW01",
    "fqbn": "esp8266:esp8266:cw01"
  },
  {
    "name": "4D Systems gen4 IoD Range",
    "fqbn": "esp8266:esp8266:gen4iod"
  },
  {
    "name": "NodeMCU 0.9 (ESP-12 Module)",
    "fqbn": "esp8266:esp8266:nodemcu"
  },
  {
    "name": "DOIT ESP-Mx DevKit (ESP8285)",
    "fqbn": "esp8266:esp8266:espmxdevkit"
  },
  {
    "name": "Digistump Oak",
    "fqbn": "esp8266:esp8266:oak"
  },
  {
    "name": "ESPDuino (ESP-13 Module)",
    "fqbn": "esp8266:esp8266:espduino"
  },
  {
    "name": "LOLIN(WEMOS) D1 ESP-WROOM-02",
    "fqbn": "esp8266:esp8266:d1_wroom_02"
  },
  {
    "name": "LOLIN(WEMOS) D1 mini Pro",
    "fqbn": "esp8266:esp8266:d1_mini_pro"
  },
  {
    "name": "Olimex MOD-WIFI-ESP8266(-DEV)",
    "fqbn": "esp8266:esp8266:modwifi"
  },
  {
    "name": "SweetPea ESP-210",
    "fqbn": "esp8266:esp8266:esp210"
  },
  {
    "name": "Generic ESP8266 Module",
    "fqbn": "esp8266:esp8266:generic"
  },
  {
    "name": "Adafruit Feather HUZZAH ESP8266",
    "fqbn": "esp8266:esp8266:huzzah"
  },
  {
    "name": "ESPectro Core",
    "fqbn": "esp8266:esp8266:espectro"
  },
  {
    "name": "LOLIN(WEMOS) D1 mini (clone)",
    "fqbn": "esp8266:esp8266:d1_mini_clone"
  },
  {
    "name": "LOLIN(WEMOS) D1 mini Lite",
    "fqbn": "esp8266:esp8266:d1_mini_lite"
  },
  {
    "name": "Phoenix 2.0",
    "fqbn": "esp8266:esp8266:phoenix_v2"
  },
  {
    "name": "SparkFun Blynk Board",
    "fqbn": "esp8266:esp8266:blynk"
  },
  {
    "name": "WifInfo",
    "fqbn": "esp8266:esp8266:wifinfo"
  },
  {
    "name": "Generic ESP8285 Module",
    "fqbn": "esp8266:esp8266:esp8285"
  },
  {
    "name": "NodeMCU 1.0 (ESP-12E Module)",
    "fqbn": "esp8266:esp8266:nodemcuv2"
  },
  {
    "name": "Phoenix 1.0",
    "fqbn": "esp8266:esp8266:phoenix_v1"
  },
  {
    "name": "LOLIN(WeMos) D1 R1",
    "fqbn": "esp8266:esp8266:d1"
  },
  {
    "name": "ESPresso Lite 2.0",
    "fqbn": "esp8266:esp8266:espresso_lite_v2"
  },
  {
    "name": "Invent One",
    "fqbn": "esp8266:esp8266:inventone"
  },
  {
    "name": "Lifely Agrumino Lemon v4",
    "fqbn": "esp8266:esp8266:agruminolemon"
  },
  {
    "name": "Seeed Wio Link",
    "fqbn": "esp8266:esp8266:wiolink"
  },
  {
    "name": "SparkFun ESP8266 Thing",
    "fqbn": "esp8266:esp8266:thing"
  },
  {
    "name": "WiFiduino",
    "fqbn": "esp8266:esp8266:wifiduino"
  },
  {
    "name": "ESPino (ESP-12 Module)",
    "fqbn": "esp8266:esp8266:espino"
  },
  {
    "name": "ITEAD Sonoff",
    "fqbn": "esp8266:esp8266:sonoff"
  },
  {
    "name": "LOLIN(WEMOS) D1 R2 & mini",
    "fqbn": "esp8266:esp8266:d1_mini"
  },
  {
    "name": "Schirmilabs Eduino WiFi",
    "fqbn": "esp8266:esp8266:eduinowifi"
  },
  {
    "name": "SparkFun ESP8266 Thing Dev",
    "fqbn": "esp8266:esp8266:thingdev"
  },
  {
    "name": "WiFi Kit 8",
    "fqbn": "esp8266:esp8266:wifi_kit_8"
  },
  {
    "name": "Arduino",
    "fqbn": "esp8266:esp8266:arduino-esp8266"
  }
]
```

@chked, what happens when you select another board? For example, `Generic ESP8266 Module` (`esp8266:esp8266:generic`). Do you still see the same error in not detecting the gdb tool path?

By the way, there is a typo: `DGB tool path`. It should be `GDB tool path`

Please take a screenshot of your boards manager while you hover over the esp8266 platform's description:

[image]

Please take a screenshot of the currently selected board from the boards list dropdown:

[image]

dankeboy36, 2024-02-22:
> @chked, what happens when you select another board? For example, `Generic ESP8266 Module` (`esp8266:esp8266:generic`). Do you still see the same error in not detecting the gdb tool path?

> Please take a screenshot of your boards manager while you hover over the esp8266 platform's description:

> Please take a screenshot of the currently selected board from the boards list dropdown:

@chked, could you please provide the following info 👆?

It would be great to see what's in the output logs. Please reference https://github.com/dankeboy36/esp-exception-decoder/issues/10#issuecomment-1727811557 on how to enable the logging. Thanks!

chked, 2024-02-22:
Thank you for reply. I got it yesterday. When I selected a ESP32 board, the error message disappeared, on every ESP8266 board, the message is still there. The reason was that the extensa...gdb.exe was not in my toolchain package. I reinstalled the toolchain package, now everything works fine. 
Perhaps, more detailed messages could help to find such problems easier.

dankeboy36, 2024-02-26:
I have installed all ESP8266 versions (`https://arduino.esp8266.com/stable/package_esp8266com_index.json`) but could not find a board with `esp12` board ID.

Generated with:

```json
{
  "Application": "arduino-cli",
  "VersionString": "0.35.3",
  "Commit": "95cfd654",
  "Status": "",
  "Date": "2024-02-19T13:24:23Z"
}
```

<details>
  <summary>Click to see all boards available from <code>esp8266:esp8266@2.1.0</code>:</summary>

```json
[
  {
    "name": "SweetPea ESP-210",
    "fqbn": "esp8266:esp8266:esp210"
  },
  {
    "name": "ThaiEasyElec's ESPino",
    "fqbn": "esp8266:esp8266:espinotee"
  },
  {
    "name": "ESPino (ESP-12 Module)",
    "fqbn": "esp8266:esp8266:espino"
  },
  {
    "name": "Generic ESP8266 Module",
    "fqbn": "esp8266:esp8266:generic"
  },
  {
    "name": "NodeMCU 0.9 (ESP-12 Module)",
    "fqbn": "esp8266:esp8266:nodemcu"
  },
  {
    "name": "Olimex MOD-WIFI-ESP8266(-DEV)",
    "fqbn": "esp8266:esp8266:modwifi"
  },
  {
    "name": "SparkFun ESP8266 Thing",
    "fqbn": "esp8266:esp8266:thing"
  },
  {
    "name": "ESPresso Lite 1.0",
    "fqbn": "esp8266:esp8266:espresso_lite_v1"
  },
  {
    "name": "NodeMCU 1.0 (ESP-12E Module)",
    "fqbn": "esp8266:esp8266:nodemcuv2"
  },
  {
    "name": "ESPresso Lite 2.0",
    "fqbn": "esp8266:esp8266:espresso_lite_v2"
  },
  {
    "name": "SparkFun ESP8266 Thing Dev",
    "fqbn": "esp8266:esp8266:thingdev"
  },
  {
    "name": "WeMos D1 R2 & mini",
    "fqbn": "esp8266:esp8266:d1_mini"
  },
  {
    "name": "WeMos D1(Retired)",
    "fqbn": "esp8266:esp8266:d1"
[... 3134 lines cut ...]
  },
  {
    "name": "SparkFun Blynk Board",
    "fqbn": "esp8266:esp8266:blynk"
  },
  {
    "name": "ThaiEasyElec's ESPino",
    "fqbn": "esp8266:esp8266:espinotee"
  },
  {
    "name": "WiFi Kit 8",
    "fqbn": "esp8266:esp8266:wifi_kit_8"
  },
  {
    "name": "LOLIN(WEMOS) D1 mini Lite",
    "fqbn": "esp8266:esp8266:d1_mini_lite"
  },
  {
    "name": "SparkFun ESP8266 Thing Dev",
    "fqbn": "esp8266:esp8266:thingdev"
  }
]
```

</details>