dankeboy36/esp-exception-decoder #16: ESP Exception Decoder Sketch path is not set

pschatzmann, 2024-02-25:
I am getting the following errors: 
ESP Exception Decoder
Sketch path is not set

dankeboy36, 2024-02-26:
Thanks for trying the decoder and reporting the bug. What's your Arduino IDE version?

pschatzmann, 2024-02-26:
Version is  2.3.2
In the Preferences I change the SketchBookLocation to /home/pschatzmann/Development/Arduino

dankeboy36, 2024-02-26:
> In the Preferences I change the SketchBookLocation to /home/pschatzmann/Development/Arduino

I could not reproduce it with 2.3.2 by changing the sketchbook location. I have to ask for some additional details from you:
 - Can you reproduce the defect by changing the sketchbook location back to the default one? That is `~/Documents/Arduino` in Linux.
 - Can you compile (verify) that sketch in Arduino IDE?
 - Can you reproduce the bug with an older version of IDE2? For example, the `2.2.1` download links for Linux are the following:
    - https://downloads.arduino.cc/arduino-ide/arduino-ide_2.2.1_Linux_64bit.AppImage
    - https://downloads.arduino.cc/arduino-ide/arduino-ide_2.2.1_Linux_64bit.zip
 - Could you please enable the logs (see https://github.com/dankeboy36/esp-exception-decoder/issues/10#issuecomment-1727811557 for the steps) and share them here?

Thank you

pschatzmann, 2024-02-26:
Strange, I activated the log and now it has started to work. 
So I guess pasting any logs will not provide any additional information.
I am closing the issue...

closed 2024-02-26 by pschatzmann

dankeboy36, 2024-02-26:
Thanks for the update. If you can reproduce it, please let me know.

mantolov, 2024-11-14:
I get this error in IDE version 2.3.3 for Windows (11).  My sketches are on a personal OneDrive.  The error doesn't show up straight away, it usually takes a while after starting up the IDE (although sometimes straight away).  Usually fixed by restarting the IDE, but then I have to wait for "package_esp32_index.json" to very slowly load (faster if I close the Serial Monitor)...  When it works, ESP Exception Decoder, works well...

kittaakos, 2024-11-14:
> My sketches are on a personal OneDrive.

It's a known defect in the Arduino IDE: https://github.com/arduino/arduino-ide/issues/1797. This decoder extension cannot do anything to work around the bug.

> Usually fixed by restarting the IDE, but then I have to wait for "package_esp32_index.json" to very slowly load (faster if I close the Serial Monitor)

This was a bug in the Arduino CLI (https://github.com/arduino/arduino-cli/issues/2719) and has been fixed, but the fixes are not yet available from the Arduino IDE 2.3.3 or the nightly builds. (I could not find the corresponding Arduino IDE GitHub issue.) The decoder extension cannot fix this issue.

https://github.com/arduino/arduino-ide/blob/63e9dfd7f5949c77ffbfb6175e2e9c2991317bca/arduino-ide-extension/package.json#L172-L174

> ESP Exception Decoder, works well...

I am glad to hear that.

Please update this issue if you can reproduce the defect after moving your sketch from OneDrive to your local disk, and closing the serial monitor.