dankeboy36/esp-exception-decoder #17: Error Activating Extension

dettmerc, 2024-02-29:
Arduino IDE 2.3.3-nightly-20230229

After creating the plugins folder and moving the folder into it, I receive the following error:

Activating extension 'ESP Exception Decoder' failed: Cannot find module '/Users/dettmerc/.arduinoIDE/plugins/esp-exception-decoder-main/dist/extension.js' Require stack: - /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/plugin-host.js

dankeboy36, 2024-02-29:
Thanks for reporting it. What's inside your `plugins` folder? Please open a _Terminal_, execute the following, and paste the output here:

```sh
ls -al ~/.arduinoIDE/plugins
```

dettmerc, 2024-02-29:
dettmerc@Christophers-MacBook-Pro ~ % ls -al ~/.arduinoIDE/plugins
total 16
drwxr-xr-x   4 dettmerc  staff   128 Feb 29 11:24 .
drwxr-xr-x@ 14 dettmerc  staff   448 Feb 29 09:20 ..
-rw-r--r--@  1 dettmerc  staff  6148 Feb 29 11:24 .DS_Store
drwxrwxr-x@ 19 dettmerc  staff   608 Jul 12  2023 esp-exception-decoder-main

dankeboy36, 2024-03-02:
Thanks for the update. I do not think this is a problem with the decoder extension itself but how you have copied it into the `plugins` folder.

This is how your `plugins` folder should look like. It contains only the extensions (`*.vsix`):

```
akos.kitta@prprpr ~ % ls -al ~/.arduinoIDE/plugins
total 6064
drwxr-xr-x   6 akos.kitta  staff      192 Dec 11 18:10 .
drwxr-xr-x  14 akos.kitta  staff      448 Sep 19 18:24 ..
-rw-r--r--@  1 akos.kitta  staff     6148 Dec 11 18:09 .DS_Store
-rw-------@  1 akos.kitta  staff  2867691 Sep 28 16:48 Arduino-helloworld-0.0.1.vsix
-rw-r--r--   1 akos.kitta  staff   103023 Dec  2 09:32 esp-exception-decoder-1.0.2.vsix
-rw-r--r--@  1 akos.kitta  staff   117168 Sep 26 17:11 pico-littlefs-upload-0.0.2.vsix
```

If you want to fix it, delete the `esp-exception-decoder-main` folder from a terminal:

> ⚠️ The command will delete the `esp-exception-decoder-main` folder with all its content from your drive without asking you! Make a backup copy if you have anything valuable inside this folder.

```
rm -rf ~/.arduinoIDE/plugins/esp-exception-decoder-main
```

Please re-download the VSIX from the release page and put it inside the `plugins` folder. The steps are in the readme: https://github.com/dankeboy36/esp-exception-decoder?tab=readme-ov-file#installation

I am closing this issue, but I am here if you need more help. You can also use the official Arduino IDE forum for support: https://forum.arduino.cc/c/software/ide-2-x/93.

closed 2024-03-02 by dankeboy36