dankeboy36/esp-exception-decoder #18: Unable to add this extension

Murugesh-Hobbyist, 2024-03-16:
getting the error,
Activating extension 'ESP Exception Decoder' failed: Cannot find module 'c:\Users\murug.arduinoIDE\plugins\esp-exception-decoder-1.0.2\dist\extension.js' Require stack: - C:\Program Files\Arduino IDE\resources\app\lib\backend\plugin-host.js

[image]

dankeboy36, 2024-03-16:
> Activating extension 'ESP Exception Decoder' failed: Cannot find module 'c:\Users\murug.arduinoIDE\plugins\esp-exception-decoder-1.0.2\dist\extension.js' Require stack: - C:\Program Files\Arduino IDE\resources\app\lib\backend\plugin-host.js

Related: https://github.com/dankeboy36/esp-exception-decoder/issues/17#issue-2161798904

----

Thanks for reporting it. I do not think this is a problem with the extension, but the way you copied the `.vsix` file to the `plugins` folder.

Please clean up the `plugins` folder and follow the installation instructions.

You can ask for help in the Arduino forum: https://forum.arduino.cc/c/software/ide-2-x/93

closed 2024-03-16 by dankeboy36

Murugesh-Hobbyist, 2024-03-17:
Thanks mate! My bad, I was copied the .zip instead of .vsix now it works great!