dankeboy36/esp-exception-decoder #21: Could not recognize stack trace/backtrace

annndruha, 2024-06-09:
Hi, while i debug my `NodeMCU v1.0` i have a error with stacktrace, i download extension but it seems like it cannot recognize the error format.

Exception Decoder log:

```text
ESP Exception Decoder
Sketch: NodeMCU_BPM180_DHT11 FQBN: esp8266:esp8266:nodemcuv2

ctx: cont
sp: 3ffffd90 end: 3fffffd0 offset: 0150
3ffffee0:  00000000 3fff0268 3ffee8a4 40205924  
3ffffef0:  3ffee7d8 3ffefe48 3ffefa5c 4020376b  
3fffff00:  00000000 00000000 00000000 3ffeead0  
3fffff10:  3fffdad0 3ffee7d8 3ffefa5c 40201ac9  

Could not recognize stack trace/backtrace

Paste exception to decode...
```

I tried different copy+paste segments of full exception text, but it seems like cut off exception to fixed lines.

**Can the extension recognize this format?**

```text
3fffff60:  00000000 00000000 3ffee780 3ffeead0  
3fffff70:  3fffdad
WiFi connected
IP address: 
192.168.1.62
Server listening

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (6):
epc1=0x40205704 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffd0 offset: 0150
3ffffee0:  00000000 3ffee8a4 0006d686 40205968  
3ffffef0:  3fffdad0 00000001 3ffefac4 40203737  
3fffff00:  40100164 00000000 00000000 3ffeead0  
3fffff10:  3fffdad0 3ffee7d8 3ffefac4 40201ac9  
3fffff20:  00000000 3ffee7d8 3ffee780 40203348  
3fffff30:  3fffdad0 00000000 00000000 002f002f  
3fffff40:  00000000 00000000 3ffe85e8 3ffee7b4  
3fffff50:  00000000 00001a7f 00000000 00000001  
3fffff60:  3fffdad0 3ffee7b4 00000000 3ffeead0  
3fffff70:  3fffdad0 3ffee7b4 3ffee780 40203b56  
3fffff80:  00000000 00000000 00001388 4020afbc  
3fffff90:  00000000 00000000 3ffe85e8 3ffeead0  
3fffffa0:  3fffdad0 00000000 3ffeeaa4 3ffeead0  
3fffffb0:  3fffdad0 00000000 3ffeeaa4 40207e18  
3fffffc0:  feefeffe feefeffe 3fffdab0 40101095  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

kittaakos, 2024-06-12:
Thanks for the quality bug report. I will look into your stack trace this week. Could you share your sketch code or its simplified version that produces the stack trace?

> this week.

Of course, this can be a bug in this decoder extension, but for now, you might want to look into https://github.com/dankeboy36/esp-exception-decoder/issues/12.

annndruha, 2024-06-12:
Unfortunately i can't reproduce this error. This error occurs once a month uptime and disappears after several resets/boots.

Last time when I see it after boot, code was simple enough: only ESP8266WebServer.h begin with no listeners: [NodeMCU_simple.ino.txt](https://github.com/user-attachments/files/15804419/NodeMCU_simple.ino.txt)

Full code just in case: [NodeMCU_BPM180_DHT11.ino.txt](https://github.com/user-attachments/files/15804363/NodeMCU_BPM180_DHT11.ino.txt)

annndruha, 2024-06-12:
**I was able to reproduce the error with a smaller traceback!**

Because of the server layer, I didn’t find error right away. The error was in this function
```cpp
float value = bmp.readTemperature();
```
After move this function in simple loop block:
```cpp
void loop() {
  auto value = bmp.readTemperature();
  Serial.println(value);
}
```
I have a smaller traceback
```text
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (6):
epc1=0x40203ea4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe40 end: 3fffffd0 offset: 0150
3fffff90:  3fffdad0 00000004 3ffee944 402040c4  
3fffffa0:  3fffdad0 00000000 3ffeeacc 402011fe  
3fffffb0:  3fffdad0 00000000 3ffeeb44 40205e28  
3fffffc0:  feefeffe feefeffe 3fffdab0 40101095  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

which was successfully decoded:

```text
Exception 6: IntegerDivideByZero: QUOS, QUOU, REMS, or REMU divisor operand is zero
PC: 0x40203ea4: Adafruit_BMP085::computeB5(int) at ...
EXCVADDR: 0x00000000

Decoding stack results
0x402040c4: Adafruit_BMP085::readTemperature() at ...
0x402011fe: loop() at ...
0x40205e28: loop_wrapper() at ...
```

WelliSolutions, 2024-10-28:
I have a similar issue. No matter what I copy/paste, it does not recognize the stack trace. It would be nice if the extension could output some progress, like

```plain
successfully detected the cut marker start
successfully detected the cut marker end
successfully detected the exception context
successfully detected the stack trace marker start
successfully detected the stack trace marker end
[...]
```

or, in an error case

```plain
missing the cut marker start
missing the cut marker end
missing the stack trace marker start
missing the stack trace marker end
...
```

In my case, I think the bug is in Arduino IDE, because it only copies what is currently visible on the screen, and not what actually has been selected. That's a pity.