dankeboy36/esp-exception-decoder #24: exception decoder 1.0.2 not showing in menu

Patsjemoe, 2024-08-04:
Hi,
I downloaded esp-exeption-decoder-1.0.2.vsix and placed it under c:\windows\users\username\.arduinoIIDE\plugins, as described in the installation guide.
I startup my arduino IDE 2.3.2, but do not see de decoder in the tools part.
Any suggestions to verify  what is wrong ?
Do I need to activate something more in arduino IDE to activate plugins maybe ?
thank you in advance.
regards
Ludo

per1234, 2024-08-04:
Hi @Patsjemoe.

> do not see de decoder in the tools part.

This is expected. Unlike Arduino IDE 1.x, Arduino IDE 2.x does not currently provide the capability for extensions to add menu items under the "**Tools**" menu (https://github.com/arduino/arduino-ide/issues/2135).

This is why you must use the "Command Palette" to open the decoder. Read the usage instructions in the documentation to learn how to do that:

https://github.com/dankeboy36/esp-exception-decoder#usage

If you have other questions about using the extension, you can post over on Arduino Forum and we'll be happy to provide assistance over there:

https://forum.arduino.cc/c/software/ide-2-x/93

closed 2024-08-04 by per1234

Patsjemoe, 2024-08-04:
> "*Tools*" menu (arduino/arduino-ide#2135

Dankeboy,
Thank you, irt works...
Regards
Ludo