dankeboy36/esp-exception-decoder #28: Plugin cant find by compiled elf file

The3ven, 2024-09-21:
ESP Exception Decoder
Sketch: NDDB_DPMCU_Project FQBN: esp32:esp32:esp32s3

The summary of the previous compilation is unavailable. Compile the sketch

dankeboy36, 2024-09-23:
Hello, thanks for trying the decoder extension.

> Compile the sketch

Have you tried to compile the sketch?

BartolomeyKant, 2025-02-25:
Hello!. Want to try this extension to find there bug is occurs.
 
Have the same issue.
```
ESP Exception Decoder
Sketch: Cloud FQBN: esp32:esp32:esp32s3

The summary of the previous compilation is unavailable. Compile the sketch
```

I compiled and recompiled it many times. Also close and reopen exception decoder terminal. Also close and reopen Arduino Ide itself and repeat all previous steps.

kittaakos, 2025-02-25:
> Hello!. Want to try this extension

❤  

Please enable logging of the VS Code Arduino API extension, restart the IDE, compile the sketch, switch to the correct output channel, and share the results here. You can find the details here: https://github.com/dankeboy36/esp-exception-decoder/issues/10#issuecomment-1727811557.

If you run into any issues, just let me know. Thanks for your support

BartolomeyKant, 2025-02-25:
Board selection
<details>
<summary>output</summary>

Updated 'boardDetails': {"buildProperties":{"version":"3.0.7","tools.esp32-arduino-libs.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a","tools.xtensa-esp32-elf-gcc.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esp-x32/2302","tools.xtensa-esp32s2-elf-gcc.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esp-xs2/2302","tools.xtensa-esp32s3-elf-gcc.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esp-xs3/2302","tools.xtensa-esp-elf-gdb.path":"/home/bartolomey/.arduino15/packages/esp32/tools/xtensa-esp-elf-gdb/12.1_20231023","tools.riscv32-esp-elf-gcc.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esp-rv32/2302","tools.riscv32-esp-elf-gdb.path":"/home/bartolomey/.arduino15/packages/esp32/tools/riscv32-esp-elf-gdb/12.1_20231023","tools.esptool_py.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esptool_py/4.6","tools.esptool_py.cmd":"esptool.py","tools.esptool_py.cmd.windows":"esptool.exe","tools.esptool_py.network_cmd":"python3 \"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7/tools/espota.py\" -r","tools.esptool_py.network_cmd.windows":"\"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7\\tools\\espota.exe\" -r","tools.esp_ota.cmd":"python3 \"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7/tools/espota.py\" -r","tools.esp_ota.cmd.windows":"\"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7\\tools\\espota.exe\" -r","tools.gen_esp32part.cmd":"python3 \"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7/tools/gen_esp32part.py\"","tools.gen_esp32part.cmd.windows":"\"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7\\tools\\gen_esp32part.exe\"","tools.gen_insights_pkg.cmd":"python3 \"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7\"/tools/gen_insights_package.py","tools.gen_insights_pkg.cmd.windows":"\"/home/bartolomey/.arduino15/packages/esp32/hardware/esp32/3.0.7\\tools\\gen_insights_package.exe\"","compiler.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esp-xs3/2302/bin/","compiler.prefix":"xtensa-esp32s3-elf-","compiler.sdk.path":"/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3","compiler.sdk.path.windows":"/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a\\esp32s3","compiler.optimization_flags":"-Os","compiler.optimization_flags.release":"-Os","compiler.optimization_flags.debug":"-Og -g3","compiler.warning_flags":"-w","compiler.warning_flags.none":"-w","compiler.warning_flags.default":"","compiler.warning_flags.more":"-Wall","compiler.warning_flags.all":"-Wall -Wextra","compiler.common_werror_flags":"-Werror=return-type","compiler.cpreprocessor.flags":"\"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/defines\" \"-I{build.source.path}\" -iprefix \"/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/include/\" \"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/includes\" \"-I/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/qio_qspi/include\"","compiler.c.flags":"\"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/c_flags\" -w -Os -Werror=return-type","compiler.cpp.flags":"\"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/cpp_flags\" -w -Os -Werror=return-type","compiler.S.flags":"\"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/S_flags\" -w -Os","compiler.c.elf.flags":"\"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/ld_flags\" \"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/ld_scripts\"","compiler.c.elf.libs":"\"@/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/flags/ld_libs\"","compiler.ar.flags":"cr","compiler.c.cmd":"xtensa-esp32s3-elf-gcc","compiler.cpp.cmd":"xtensa-esp32s3-elf-g++","compiler.S.cmd":"xtensa-esp32s3-elf-gcc","compiler.c.elf.cmd":"xtensa-esp32s3-elf-g++","compiler.as.cmd":"xtensa-esp32s3-elf-as","compiler.ar.cmd":"xtensa-esp32s3-elf-gcc-ar","compiler.size.cmd":"xtensa-esp32s3-elf-size","compiler.c.extra_flags":"-MMD -c","compiler.cpp.extra_flags":"-MMD -c","compiler.S.extra_flags":"-MMD -c -x assembler-with-cpp","compiler.c.elf.extra_flags":"\"-Wl,--Map={build.path}/{build.project_name}.map\" \"-L/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/lib\" \"-L/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/ld\" \"-L/home/bartolomey/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s3/qio_qspi\" \"-Wl,--wrap=esp_panic_handler\"","compiler.ar.extra_flags":"","compiler.objcopy.eep.extra_flags":"","compiler.elf2hex.extra_flags":"","compiler.libraries.ldflags":"","build.extra_flags.esp32":"-DARDUINO_USB_CDC_ON_BOOT=0","build.extra_flags.esp32s3":"-DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_ [... cut ...]
      "selected": false,
      "missing": false
    },
    "arduino+serial:///dev/ttyACM3": {
      "name": "ESP32S3 Dev Module",
      "fqbn": "esp32:esp32:esp32s3",
      "packageId": {
        "vendorId": "esp32",
        "arch": "esp32"
      },
      "packageName": "esp32",
      "manuallyInstalled": false,
      "selected": false,
      "missing": false
    }
  }
}

</details>

But I don't get how to open `ESP Exception Decoder (Log)`  I can see onlu the `Arduino Api` channel in `Show Output Channel` menu.

kittaakos, 2025-02-25:
Please do:
 - start IDE and select the board if not selected,
 - open the Arduino API output channel,
 - clear the content,
 - verify,
 - open the Arduino API output channel and
 - share the `compileSummary`. 🙏 

https://github.com/user-attachments/assets/63371a14-3de4-4764-ade0-4208fa6104e6

BartolomeyKant, 2025-02-25:
It's empty!
I've got `Sketch uses...` thing and switched to `Arduino Api` channel, and it's empty.

kittaakos, 2025-02-25:
> channel, and it's empty.

If it's not set by the IDE, at least we know that the decoder exception behaves correctly.

There could be two main reasons why it's not set:
 - after the sketch compilation, the IDE [executes the `arduino.languageserver.notifyBuildDidComplete` command](https://github.com/arduino/arduino-ide/blob/6eef09efd8f49827cabb4eabd8cb2ff99ffae09a/arduino-ide-extension/src/node/core-service-impl.ts#L184) to send the sketch build details to the language server (the language server is a different component in IDE)
 - the response object has an invalid shape ([See in the Arduino IDE source](https://github.com/arduino/arduino-ide/blob/6eef09efd8f49827cabb4eabd8cb2ff99ffae09a/arduino-ide-extension/src/browser/contributions/update-arduino-state.ts#L66))

Please start the IDE from a terminal, clear the terminal, and run a verification if you see something like this or an error that the command execution has failed:
```
2025-02-25T11:53:01.113Z root INFO Executing 'arduino.languageserver.notifyBuildDidComplete' with "file:///Users/kittaakos/Library/Caches/arduino/sketches/EC49C9101B3596EEB249E6FCAF68E688"
```

BartolomeyKant, 2025-02-25:
> Please start the IDE from a terminal, clear the terminal, and run a verification if you see something like this or an error that the command execution has failed:

Yes! There is error!
```
2025-02-25T12:21:34.013Z root INFO Executing 'arduino.languageserver.notifyBuildDidComplete' with "file:///home/bartolomey/.cache/arduino/sketches/E451323152A4E58EABD2AB4E5B3EE5E4"
2025-02-25T12:21:34.045Z root ERROR Request executeCommand failed with error: Language client is not ready yet Error: Language client is not ready yet
```

I had an error with language server start before due to missing arduino-language-server, and  I installed it, and error gone. But it still doesn't work.

kittaakos, 2025-02-25:
Thanks for checking. It's not a bug in the decoder extension.

> and I installed it,

How did you install it? It should be part of the IDE installation:

From macOS:
```
 % /Applications/Arduino\ IDE.app/Contents/Resources/app/lib/backend/resources/arduino-language-server
13:36:32.897577 Path to ArduinoCLI config file must be set.
```

Do you have the Arduino Language Server available from `app/lib/backend/resources` inside your IDE folder?

It may not be the best IDE for listening to command executions and populating the Arduino state for Arduino IDE extensions. If you're interested in trying out a proposed fix, I'll open a PR and ping you there. Let me know if you're willing to assist me.

BartolomeyKant, 2025-02-25:
I'm on manjaro linux.

Installed it by `yay -Su arduino-language-server` - it's a separate package there.

It has this file list
```bash
yay -Ql arduino-language-server
arduino-language-server /usr/
arduino-language-server /usr/bin/
arduino-language-server /usr/bin/arduino-language-server
arduino-language-server /usr/share/
arduino-language-server /usr/share/doc/
arduino-language-server /usr/share/doc/arduino-language-server/
arduino-language-server /usr/share/doc/arduino-language-server/README.md
```

And arduino ide itself
```bash
yay -Ql arduino-ide | grep language-server
arduino-ide /usr/lib/arduino-ide/lib/backend/resources/arduino-language-server
```
Which is
```bash
ls -l /usr/lib/arduino-ide/lib/backend/resources/arduino-language-server
lrwxrwxrwx 1 root root 32 дек  3 23:12 /usr/lib/arduino-ide/lib/backend/resources/arduino-language-server -> /usr/bin/arduino-language-server*
```

dankeboy36, 2025-02-27:
@BartolomeyKant, thanks for helping with the investigation.

I created a fix in the IDE; please help with the verification.

Please reference the documentation of how to use the IDE tester builds: https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md.

You will find the PR at https://github.com/arduino/arduino-ide/pull/2643.

BartolomeyKant, 2025-03-04:
> I created a fix in the IDE; please help with the verification.

I'm sorry for late answer. I've tested it, and it works pretty well as expected.

dankeboy36, 2025-03-04:
> I'm sorry for late answer. 

Thank you so much for verifying the build and getting back with the update.

> I'm sorry for late answer. I've tested it, and it works pretty well as expected.

I am glad the fix works.

> Do you need additional logs, or should I search for something on it myself?

No, there is nothing to do on your side. Once the Arduino devs merge https://github.com/arduino/arduino-ide/pull/2643, you can get the fix with the following day's nightly Arduino IDE build (https://www.arduino.cc/en/software#nightly-builds) and can use the decoder.

dankeboy36, 2025-03-11:
The Arduino team has added the [fix](https://github.com/arduino/arduino-ide/commit/5ec191500028582e687c950b78966e4b7aefaad8) to the nightly version of the IDE: https://www.arduino.cc/en/software#nightly-builds. 

Please use this nightly build until the next official IDE release comes out. 

I will mark this issue as fixed, but if you still have problems, let me know.

closed 2025-03-11 by dankeboy36