dankeboy36/esp-exception-decoder #31: The decoder does not recognize the stack trace after copy/pasting to the terminal

dankeboy36, 2024-10-28:
I have a similar issue. No matter what I copy/paste, it does not recognize the stack trace. It would be nice if the extension could output some progress, like

```plain
successfully detected the cut marker start
successfully detected the cut marker end
successfully detected the exception context
successfully detected the stack trace marker start
successfully detected the stack trace marker end
[...]
```

or, in an error case

```plain
missing the cut marker start
missing the cut marker end
missing the stack trace marker start
missing the stack trace marker end
[...]
```

In my case, I think the bug is in Arduino IDE, because it only copies what is currently visible on the screen, and not what actually has been selected [[probably this bug]](https://github.com/arduino/arduino-ide/issues/1081). That's a pity. My traceback is several dozens of lines long.

Also, 

```plain
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```
is not very helpful, because I don't know whether this is the start or the end of the cut.

_Originally posted by @WelliSolutions in https://github.com/dankeboy36/esp-exception-decoder/issues/21#issuecomment-2441674329_

dankeboy36, 2024-10-28:
Thanks for using the extension.

> I have a similar issue. No matter what I copy/paste, it does not recognize the stack trace

@WelliSolutions, please copy/paste the content instead of trying to guess which Arduino IDE bug truncates the monitor content. Otherwise, it's impossible to give support.

> Also,
> 
> ```
> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
> ```
> 
> is not very helpful, because I don't know whether this is the start or the end of the cut.>

The board's stack trace output produces this; how can the decoder extension change your board's output?

WelliSolutions, 2024-10-31:
> please copy/paste the content instead

Well, I tried to, but it is not possible ...

> instead of trying to guess which Arduino IDE bug truncates the monitor content

... because of that Arduino IDE bug.

> The board's stack trace output produces this;

I didn't know that.

kittaakos, 2024-11-01:
> > please copy/paste the content instead
> 
> Well, I tried to, but it is not possible ...

I understand your frustration, but you should copy/paste the content here to the GitHub issue as a code block. Otherwise, I cannot help resolving it. I hope this helps

alastaira, 2024-11-11:
I ended up here after a Google search, having discovered the same problem and can propose an (imperfect) solution.

As hinted in the first post, this is not a bug in the exception decoder, but in the Arduino IDE. The problem is that it is not possible to highlight and copy the entire stack trace to the clipboard, because when you Ctrl+C from the serial monitor, the only content copied is that currently visible in the Serial Monitor pane. It took me some time to discover this, but it's very easy to verify by simply attempting to paste the stack trace into, say, Notepad instead. You'll see that you only get a partial result, which is why the exception decoder therefore doesn't work.

The workaround I've used is to enlarge the Serial Monitor pane as large as possible, and then reduce the text size (Ctrl -) until the entire stack trace can be viewed at once. Then, it can be highlighted, copied, and pasted. 

[image]

WelliSolutions, 2024-11-11:
> The workaround I've used is to enlarge the Serial Monitor pane as large as possible, and then reduce the text size (Ctrl -) until the entire stack trace can be viewed at once. Then, it can be highlighted, copied, and pasted.

My stack trace is about 6x as long. It won't fit into the Serial Monitor, even on a 4K screen.

alastaira, 2024-11-11:
Oh, I agree, it's far from a perfect solution. But it's an issue for the Arduino IDE, not the ESP Exception Decoder.

per1234, 2024-11-12:
> It won't fit into the Serial Monitor

You can get a bit more vertical height to fit the stack trace by moving Serial Monitor to the right hand side panel of the Arduino IDE window:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
   A menu will appear on the editor toolbar:
   [image]
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
   **ⓘ** You can scroll down through the list of commands to find it or type the name in the field.
   A "**Preferences**" tab will open in the Arduino IDE main panel.
1. Type `arduino.monitor.dockPanel` in the "**Search Settings**" field of the "**Preferences**" tab.
1. Select "**right**" from the menu under the "**Arduino › Monitor: Dock Panel**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

If that still isn't enough, you can use a standalone serial terminal instead of Serial Monitor as a workaround for this bug in Serial Monitor. I can recommend the open source PuTTY:

https://www.chiark.greenend.org.uk/~sgtatham/putty/

When using an external terminal, make sure to close the serial port in the terminal before uploading a sketch or using Serial Monitor. The reason is that only one application can have the port open at a time so the external terminal would block Arduino IDE from using the port. Serial Monitor is smart enough to automatically juggle the control over the port during an upload so we don't have to do this manually when using Serial Monitor.

> it's an issue for the Arduino IDE

It is being tracked here: https://github.com/arduino/arduino-ide/issues/1081

T-S-70, 2024-11-26:
I have a similar problem but this has nothing to do with the copy/paste. The decoder reports "Could not recognize stack trace/backtrace" after pasting the entire dump from the serial stack trace.

Board is ESP32-C3 (XIAO_ESP32C3). I also tried the Ardu Uno Rev 4 (that's also ESP32 based) but that comes back with a "Unsupported board: 'arduino:renesas_uno:unor4wifi'" error. 2 Conclusions: 1) the C3 seems supported (but does not work); 2) the Uno R4 should also be added since that would be the same core.

For pasting the stack I tried a number of things, paste only the stack trace without the registry dump, paste without the top error line, paste without the ELF SHA line and more, none of those worked.

### For repro here's the code:
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("SETUP");
}

class a
{
public:
  int a = 323;
  int geta() {return a;}
};

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("LOOP");
  delay(2000);
  a* _a = 0;
  //a* _a = new a();
  Serial.println(_a->geta());
}
```

### stack trace:
```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42000074  RA      : 0x42000072  SP      : 0x3fc94f70  GP      : 0x3fc8c000  
TP      : 0x3fc8830c  T0      : 0x4005890e  T1      : 0x18000000  T2      : 0x00000000  
S0/FP   : 0x3fc8d000  S1      : 0x00000000  A0      : 0x00000001  A1      : 0x00000001  
A2      : 0x0000000a  A3      : 0x00000004  A4      : 0x600c0000  A5      : 0x00000000  
A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x3fc95480  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc94f70: 0x00000000 0x00000000 0x00000000 0x4200360a 0x00000000 0x00000000 0x00000000 0x403872d8
3fc94f90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94fb0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234
3fc94fd0: 0x00000154 0x3fc94ed0 0x000007d6 0x3fc8d0e8 0x3fc8d0e8 0x3fc94fd4 0x3fc8d0e0 0x00000018
3fc94ff0: 0x5c7915e7 0x7b4e7a8c 0x3fc94fd4 0x00000000 0x00000001 0x3fc92fc4 0x706f6f6c 0x6b736154
3fc95010: 0xcfb6bc00 0x0093573e 0x00000000 0x3fc94fc0 0x00000001 0x00000000 0x00000000 0x00000000
3fc95030: 0x00000000 0x3fc8ed50 0x3fc8edb8 0x3fc8ee20 0x00000000 0x00000000 0x00000001 0x00000000
3fc95050: 0x00000000 0x00000000 0x4201a80c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95070: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc950b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc950d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc950f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc95110: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x02000000 0xbaad5678 0x00000060
3fc95130: 0xabba1234 0x00000054 0x00000000 0x3fc95138 0x00000000 0x00000000 0x00000000 0x3fc95150
3fc95150: 0xffffffff 0x3fc95150 0x3fc95150 0x00000000 0x3fc95164 0xffffffff 0x3fc95164 0x3fc95164
3fc95170: 0x00000001 0x00000001 0x00000000 0x0100ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678
3fc95190: 0x00000160 0xabba1234 0x00000154 0x3fc951f0 0x3fc951f0 0x3fc952f0 0x3fc952ef 0x00000000
3fc951b0: 0x3fc951b4 0xffffffff 0x3fc951b4 0x3fc951b4 0x00000000 0x3fc951c8 0xffffffff 0x3fc951c8
3fc951d0: 0x3fc951c8 0x00000000 0x00000100 0x00000001 0xdf00ffff 0x00000000 0xb33fffff 0x00000000
3fc951f0: 0x1b5a20c3 0xee7e423f 0x658fe1be 0x6dbb0365 0x2768218d 0x1f90cd68 0x182c00b2 0x4294b268
3fc95210: 0xa04df2fd 0x75048135 0x03eb0af3 0xeb7eda73 0x24338fa1 0xf7ed1cd7 0x8e5fc680 0x1979b0ae
[... 48 lines cut ...]
3fc95130: 0xabba1234 0x00000054 0x00000000 0x3fc95138 0x00000000 0x00000000 0x00000000 0x3fc95150
3fc95150: 0xffffffff 0x3fc95150 0x3fc95150 0x00000000 0x3fc95164 0xffffffff 0x3fc95164 0x3fc95164
3fc95170: 0x00000001 0x00000001 0x00000000 0x0100ffff 0x00000000 0xb33fffff 0x00000000 0xbaad5678
3fc95190: 0x00000160 0xabba1234 0x00000154 0x3fc951f0 0x3fc951f0 0x3fc952f0 0x3fc952ef 0x00000000
3fc951b0: 0x3fc951b4 0xffffffff 0x3fc951b4 0x3fc951b4 0x00000000 0x3fc951c8 0xffffffff 0x3fc951c8
3fc951d0: 0x3fc951c8 0x00000000 0x00000100 0x00000001 0xdf00ffff 0x00000000 0xb33fffff 0x00000000
3fc951f0: 0x1b5a20c3 0xee7e423f 0x658fe1be 0x6dbb0365 0x2768218d 0x1f90cd68 0x182c00b2 0x4294b268
3fc95210: 0xa04df2fd 0x75048135 0x03eb0af3 0xeb7eda73 0x24338fa1 0xf7ed1cd7 0x8e5fc680 0x1979b0ae
3fc95230: 0xd3ef46f1 0x6372363e 0x38a11040 0x2e8efe8a 0x6d068dd2 0x8b00a172 0xb9830d9c 0x8769eab7
3fc95250: 0xa7c8daf7 0xbbb33bb5 0xbaf6b4e8 0xbf37a935 0x486d4432 0x3d990202 0x170898bb 0x97abc2b6
3fc95270: 0x4eb7fc73 0x055c4723 0x0733dec4 0xb5069893 0x66965406 0x8d46a1ec 0x36c1315a 0xdae43046
3fc95290: 0x20715846 0x6f1f2f25 0x9bd65078 0x91c8d748 0x3fdf2491 0xd4861e1d 0x9098b0f1 0x10529681
[... cut ...]
3fc952f0: 0xbaad5678 0x0000007c 0xabba1234 0x00000070 0x00000100 0x00000100 0x00000002 0x40383d14
3fc95310: 0x40384394 0x40383c4a 0x403848dc 0x403848c0 0x3fc9538d 0x3fc9538d 0x3fc95380 0x3fc95380
3fc95330: 0x3fc95380 0x3fc95480 0x0000000d 0x00000000 0x3fc95344 0xffffffff 0x3fc95344 0x3fc95344
3fc95350: 0x00000000 0x3fc95358 0xffffffff 0x3fc95358 0x3fc95358 0x00000000 0xb33fffff 0x00000000

ELF file SHA256: d0e2e7e7f0a0afca

Could not recognize stack trace/backtrace

Paste exception to decode...
```

dankeboy36, 2024-11-26:
> Board is ESP32-C3

Please see https://github.com/dankeboy36/esp-exception-decoder/issues/6.

> but that comes back with a "Unsupported board: 'arduino:renesas_uno:unor4wifi'" error. 2 Conclusions: 1)

`arduino:renesas_uno` is unsupported by the decoder extension. These are the supported archs:

https://github.com/dankeboy36/esp-exception-decoder/blob/acecf058c145e991844e6234ede1a571af753d95/src/decoder.ts#L394-L395