dankeboy36/esp-exception-decoder #34: ESP Exception Decoder Not Working

strangeanka, 2024-11-05:
As you said, I select the plugin from the drop-down menu using the Ctrl Shift + P shortcut, but there is no change

kittaakos, 2024-11-05:
Thank you for using the decoder extension.

> I select the plugin from the drop-down menu using the Ctrl Shift + P shortcut, but there is no change

You need to provide some additional info on the problem. Otherwise, I cannot help you.

 - What's your Arduino IDE version?
 - When the Arduino IDE starts, do you see the `ESP Exception Decoder: Show Decoder Terminal` command after opening the command palette with `cmd/ctrl+shift+P`?
 - If yes, what does it print to the console?

strangeanka, 2024-11-05:
My arduino version 2.3.3

After the Ctrl + Shift + P shortcut, no tab is opened, as I said, there is no change

strangeanka, 2024-11-05:
/home/strange/2024-11-05 19-16-22.mp4

kittaakos, 2024-11-05:
> no tab is opened, as I said, there is no change

Can you take a screenshot and attach it to this issue? Thanks

strangeanka, 2024-11-05:
> > dediğim gibi sekme açılmadı, değişiklik yok
> 
> Bir ekran görüntüsü alıp bu konuya ekleyebilir misiniz? Teşekkürler

https://streamable.com/pvdo7h

dankeboy36, 2024-11-06:
Thanks for taking the time and uploading a screencast. Please attach the logs from the Arduino IDE to this issue. 

The log files location depending on your OS:

Windows:
```
C:\Users\<username>\AppData\Roaming\Arduino IDE\
```

macOS:
```
/Users/<username>/Library/Logs/Arduino IDE/
```

Linux:
```
/home/<username>/.config/Arduino IDE/
```

Let me know if you need assistance

skokovsergey, 2025-02-02:
I checked logs and found the following error. Removing this folder helped.
`root ERROR Failed to load plugin dependencies from '/private/var/folders/82/cjkmj6gx1g7c2t3l8dmrjpq40000gn/T/vscode-unpacked/esp-exception-decoder-1.0.2.vsix' path Error: ENOENT: no such file or directory, open '/private/var/folders/82/cjkmj6gx1g7c2t3l8dmrjpq40000gn/T/vscode-unpacked/esp-exception-decoder-1.0.2.vsix/package.json'`

dankeboy36, 2025-02-02:
> I checked logs and found the following error. Removing this folder helped.

I’m glad you figured it out and shared the solution. For anyone experiencing the same error, please see: https://github.com/dankeboy36/esp-exception-decoder/issues/17#issuecomment-1974804485. I’m just guessing here, though.

dagnall53, 2025-03-02:
@dankeboy36 
I'm also having trouble getting the exception decoder to "work". My issue seems to be that my Arduino (2.3.4) (windows) does not seem to have the expected directory structure ?? - 
I have only these sub directories for Arduino IDE: 
locales, plugins, resources
/plugins contains ONLY the esp-exception-decoder-1.0.2.vsix file. 

I also find this vsix file in 'C:\Users\admin\.arduinoIDE\plugin-storage' and I do not think I put that there!

Searching the log(attached)  does not find an error like your previous answer.. 
But I do see some errors that apparently relate to missing ToolchainPath ???

in log from C:\Users\admin\AppData\Roaming\Arduino IDE copied below: .. I find
" root ERROR [hosted-plugin: 10212] Conflict in settings file Default: Ignoring cortex-debug.armToolchainPath.linux as cortex-debug.armToolchainPath is null"
Am I missing a tool path (or equivalent) for Version 2.3.4? Did my original upgrade try to use original directores? and is this causing issues now?
 
I keep an older Arduino IDE also installed, (so I can use Exception decoder), and that of course has its own directories:
I think my older arduino uses : C:\Program Files (x86)\Arduino\tools
C:\Users\admin\AppData\Roaming\arduino-ide etc.. 
I also have C:\Users\admin\AppData\Local\Arduino15 
Is my V2.3.4 upgrade somehow pointing to older, or missing directores? 

PS.. checking the Older ide, I also used the ESP8266FS - that was kept in tools along with the exception decoder.. Do you know if there a IDE2.3.4 version of this?? Getting that running may assist in getting the Exception decoder running.  
 

[2025-03-02_log.log](https://github.com/user-attachments/files/19042570/2025-03-02_log.log)

 Many thanks

kittaakos, 2025-03-02:
> I'm also having trouble getting the exception decoder to "work".

@dagnall53, thanks for trying this extension. I looked at the logs, and the decoder isn't mentioned. Following the [installation steps](https://github.com/dankeboy36/esp-exception-decoder?tab=readme-ov-file#installation) should help. If you run into any issues, the Arduino Forum is a great place to ask for support: https://forum.arduino.cc/c/development-tools/ide-2-x/93. 

Thanks for understanding, and happy tinkering!

dagnall53, 2025-03-02:
@kittaakos 
I thought I had followed the instructions, so to be sure and record here: 
I have deleted the visz file in both C:\Program Files\Arduino IDE\plugins and C:\Users\admin\.arduinoIDE\plugin-storage.
But have not deleted the \plugins directory...
Restarted Arduino 2.3.4 and closed it . 
Tried to download the visx direct from github to the plugins directory- Failed and it offered to save to Plugin-storage instead (Is that how it got there?) Copied to downloads and cut paste using Explore (and said YES to Admin rights copy.
Restart Arduino and STILL no ESP visible under the ctrl/shift P list. 
Checking the updated log I see :PLUGIN_HOST(5400): initializing(vscode-arduino-api@0.1.2 with C:\Program Files\Arduino IDE\resources\app\lib\backend\plugin-vscode-init)>
and a mention of PLUGIN_HOST(10212): PluginManagerExtImpl/loadPlugin(
C:\Program Files\Arduino IDE\resources\app\plugins\vscode-arduino-api\extension\dist\extension.js)..
SO I have copied the visx to THAT directory, and now I have it!!

@dankeboy36 You may wish to add a note about this alternate Plugins directory location..!