dankeboy36/esp-exception-decoder #35: Version for IDE 2.3.4?

philmay57, 2024-12-03:
Just installed 2.3.4 and decoder kept asking for the application to be rebuilt.  Backed out to a copy of IDE 2.3.2 I had downloaded, rebuilt app and the decoder then decoded the stack trace for me, but a pain to have to reinstall the old IDE for every non-obvious crash.

dankeboy36, 2024-12-03:
Thanks for using the decoder extension.

I updated Arduino IDE to 2.3.4, and I could not reproduce anything like you have reported.

[image]

> Just installed 2.3.4 and decoder kept asking for the application to be rebuilt.

Can you share a screencast or a screenshot, at least?

> I had downloaded, rebuilt app and the decoder

What did you do?

philmay57, 2024-12-04:
Hi, thanks for the prompt response.  I just reinstalled 2.3.4 and of course now it all works fine.  Apologies for wasting your time, and thanks for supporting this excellent plugin.
Regards
Phil

closed 2024-12-04 by philmay57