dankeboy36/esp-exception-decoder #41: Arduino 2.3.5  - not installing ESP

dagnall53, 2025-04-03:
I have just updated to IDE 2.3.5 from 2.3.4 where - after a lot of tests I had managed to get the decoder working - see my comments in 
https://github.com/dankeboy36/esp-exception-decoder/issues/34#issuecomment-2692811361

Unfortunately this IDE update has again disabled the ESP Exception decoder. 

Have they changed plugins directory structure again? 
When the .vsix is loaded in  C:\Program Files\Arduino IDE\resources\app\plugins
it does not work.

dankeboy36, 2025-04-03:
> see my comments in [#34 (comment)](https://github.com/dankeboy36/esp-exception-decoder/issues/34#issuecomment-2692811361)

Hello, and thank you for opening this ticket.

I have seen your remark. Thanks for the ping. However, I believe it's best to stick with one correct way of installing the decoder extension. While your method might work by accident, changing files inside the Arduino IDE can lead to problems with future updates. This is why I didn't update the documentation.

For now, I suggest following the steps from the documentation here: https://github.com/dankeboy36/esp-exception-decoder?tab=readme-ov-file#installation.

I have not tried the new Arduino IDE version, nor if the decoder runs with it, so I will keep this issue open for a while.

dagnall53, 2025-04-03:
My possible confusion between  C:\Program Files\Arduino
IDE\resources\app\plugins
and C:\Users\admin\.arduinoIDE\plugins

So I now have it only in the  C:\Users\admin\.arduinoIDE\plugins

The new IDE has the .arduinoIDE with (amongst others) these folders:
plugins and deployedPlugins.
I find after starting the IDE, that there is a
folder C:\Users\admin\.arduinoIDE\deployedPlugins\esp-exception-decoder-1.1.0
with content
[image: image.png]
and it runs.  - but FYI  the ctrl-shift-p does not seem to search for
things starting ESP as you type , and I had to scroll down to find it.
Cheers
D

dankeboy36, 2025-07-07:
> I find after starting the IDE, that there is a folder C:\Users\admin\.arduinoIDE\deployedPlugins\esp-exception-decoder-1.1.0 with content

Yes, this is expected behavior. When you place the `.vsix` file in the `plugins` folder, the Arduino IDE (which uses Theia under the hood) automatically extracts it to the `deployedPlugins` folder.

> but FYI the ctrl-shift-p does not seem to search for things starting ESP as you type , and I had to scroll down to find it.

I wasn't able to reproduce this. The decoder command shows up normally in the command palette as you type. Could you double-check if this consistently happens in your setup? Also, which version of the IDE are you running? Everything works correctly on my end with 2.3.6.

At this point, the decoder is working as expected. If this issue primarily concerns improving the installation documentation or clarifying information for future users, I'll close the issue. I don't plan to update the documentation beyond what’s already there. See why: https://github.com/dankeboy36/esp-exception-decoder/issues/41#issuecomment-2776278738.

Thank you!

dankeboy36, 2025-07-24:
> Looks like I have the same issue. Using version 1.1.1 of the decoder, and 2.3.6 of the Arduino IDE. The file it looks for (extension.js) isn't found anywhere?
> 
> `Activating extension 'ESP Exception Decoder' failed: Cannot find module '/home/peter/.arduinoIDE/plugins/esp-exception-decoder-1.1.1.vsix/dist/extension.js' Require stack: - /app/arduino-ide/resources/app/lib/backend/plugin-host.js `

Hi, thank you for trying the decoder and searching for an existing issue. I think your problem is unrelated, and something went wrong during the installation. See https://github.com/dankeboy36/esp-exception-decoder/issues/18 if it helps.

dankeboy36, 2025-07-25:
I am closing this issue. The original problem was attempting to install the decoder extension in an incorrect location and updating the documentation.

> Have they changed plugins directory structure again?
> When the .vsix is loaded in C:\Program Files\Arduino IDE\resources\app\plugins
> it does not work.

closed 2025-07-25 by dankeboy36