dankeboy36/esp-exception-decoder #42: unble to decode stack.

fanfanlatulipe26, 2025-04-29:
ESP32 C3
Arduino IDE 2.3.6
ESP32 arduino core 3.2.0
After a crash, I tried the Exception decoder (release 1.1.0) but only get the following result:

Question: is the decoder usable in Arduino IDE 1.x.y  ?.   Copy/pasting a large amount of data from Serial Monitor IDE 2.x.y is really a pain ...

```
Exception 2: Illegal instruction
MEPC: 0x403813ea
MTVAL: 0x00000000

Decoding stack results
details=0x3fc94044 <xIsrStack+1668> "essaf treliax :dueuQceReevieeuq c.eu451:!( 0 ( (saTxteGkehcSeludatSr: undefined
0x30e13840: ??
```

Here is the full data pasted in the decoder:

```
assert failed: xQueueReceive queue.c:1540 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))
Core  0 register dump:
MEPC    : 0x403813ea  RA      : 0x403876cc  SP      : 0x3fc93fe0  GP      : 0x3fc90000  
TP      : 0x3fca7a50  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000001  S1      : 0x3fc9418e  A0      : 0x3fc94044  A1      : 0x3fc924f9  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc98000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000049  S3      : 0x3fc94038  
S4      : 0x3fc94038  S5      : 0x3fc94044  S6      : 0x00000001  S7      : 0x3fce0000  
S8      : 0x3ff1b000  S9      : 0x3fce0000  S10     : 0x3fcdf934  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000 

Stack memory:
3fc93fe0: 0x00000000 0x00000000 0x3c0d31d0 0x4038e130 0x00000000 0x00000000 0x00000000 0x3fc924f8
3fc94000: 0x00000000 0x00000000 0x00000000 0x30343531 0x00000000 0x3fc92860 0x3c0d31d0 0x3fc92ca8
3fc94020: 0x3c0c5622 0x3fc92870 0x3fc9400c 0x3fc92874 0x3c0c57c8 0x3fc924f8 0x00000000 0x00000000
3fc94040: 0x00000000 0x65737361 0x66207472 0x656c6961 0x78203a64 0x75657551 0x63655265 0x65766965
3fc94060: 0x65757120 0x632e6575 0x3435313a 0x21282030 0x20282028 0x73615478 0x7465476b 0x65686353
3fc94080: 0x656c7564 0x61745372 0x29286574 0x203d3d20 0x20282028 0x65736142 0x65707954 0x2920745f
3fc940a0: 0x29203020 0x26202920 0x20282026 0x63695478 0x6f54736b 0x74696157 0x203d2120 0x20292030
3fc940c0: 0x00002929 0x00000000 0x60016400 0x00000001 0x00000008 0x3fcae8b4 0x00000000 0x3fcae920
3fc940e0: 0x3fc9413c 0x00000000 0x3fcaed00 0x40382c9a 0x00000000 0x00000000 0x00000000 0x01000000
3fc94100: 0x00000000 0x00000000 0x00000000 0x5ad19f87 0x00000001 0x3fcaed00 0x3fcae8b4 0x00000008
3fc94120: 0x3fc94178 0x00000000 0x3fcaec1c 0x403880ae 0x3fcae67c 0x000000a0 0x3fcae6a4 0xffffffff
3fc94140: 0x00000002 0xff000000 0x3fcae82c 0x40387ef8 0x00000001 0x3fcaed00 0x3fcae8b4 0x00000008
3fc94160: 0x00000000 0x3fcae904 0x3fcae920 0x4201a5f0 0x00000000 0xff000000 0x00000000 0x40387ef8
3fc94180: 0x3fce0000 0x3fca59b4 0x00000001 0x00001000 0x3fcae944 0x00000000 0x3fcae920 0x420006e6
3fc941a0: 0x3fcae944 0x00000001 0x3fcae920 0x4038339a 0x00000000 0x3fcaea4c 0x00000005 0x0000000b
3fc941c0: 0x3fce0000 0x3fca59b4 0x00000000 0x00000001 0x00001881 0x80000009 0x3fcaecd8 0x403820aa
3fc941e0: 0x00001881 0x80000004 0x3fca79ac 0x40380270 0x60023000 0x4038602e 0x4038603a 0x3fc94258
3fc94200: 0x3fc94258 0x3fc942f8 0x3fc942e8 0x00000000 0x3fc94214 0xffffffff 0x3fc94214 0x3fc94214
3fc94220: 0x00000001 0x3fc94228 0xffffffff 0x3fc9ee48 0x3fc9ee48 0x00000000 0x0000000a 0x00000010
3fc94240: 0x0001ffff 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94260: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc94280: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc942a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc942c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc942e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc94300
3fc94300: 0xffffffff 0x3fc94300 0x3fc94300 0x00000000 0x3fc94314 0xffffffff 0x3fc94314 0x3fc94314
3fc94320: 0x00000000 0x3fc94320 0x00000000 0x00000000 0x00000000 0x3fc94338 0xffffffff 0x3fc94338
3fc94340: 0x3fc94338 0x00000000 0x3fc9434c 0xffffffff 0x3fc9434c 0x3fc9434c 0x00000001 0x00000001
3fc94360: 0x00000000 0x0001ffff 0x00000000 0x00000000 0x00000004 0x00000000 0x00000000 0x00000000
3fc94380: 0x3fc9437c 0x00000000 0x00000000 0x00000000 0x3fc94394 0xffffffff 0x3fc94394 0x3fc94394
3fc943a0: 0x00000000 0x3fc943a8 0xffffffff 0x3fc943a8 0x3fc943a8 0x00000001 0x00000001 0x00000000
3fc943c0: 0x0001ffff 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x60023000 0x4038602e
```

fanfanlatulipe26, 2025-04-30:
I made a new compilation/run/crash. Here are 2 files: the .elf and the stack/registerd dump.   
I got the elf file with the IDE  Sketch/Compile binary command.   
[stack_elf.zip](https://github.com/user-attachments/files/19971961/stack_elf.zip)

kittaakos, 2025-04-30:
Thanks for the ZIP. I confirm the raw stdout from GDB shows more info than the decoder extension. The line parser must be relaxed, much more relaxed:

```
panic_abort (details=0x3fc93b54 <xIsrStack+1668> "essaf treliax :dueuQceReevieeuq c.eu451:!( 0 ( (saTxteGkehcSeludatSr)(et ==  ( (esaBepyT) t_) 0 & )  ( &ciTxoTsktiaW =!  ) 0") at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:468
#0  panic_abort (details=0x3fc93b54 <xIsrStack+1668> "essaf treliax :dueuQceReevieeuq c.eu451:!( 0 ( (saTxteGkehcSeludatSr)(et ==  ( (esaBepyT) t_) 0 & )  ( &ciTxoTsktiaW =!  ) 0") at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:468
#1  0x403872b2 in esp_system_abort (details=<optimized out>) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:87
#2  0xd6dc3840 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

However, the raw GDB output does not show the problem either. I wrote a short script to use `addr2line` from the ESP32 core (`~/Library/Arduino15/packages/esp32/tools/esp-rv32/2405/bin/riscv32-esp-elf-addr2line`), and blindly decode all addresses.

The output is still super noisy, and it's just a standalone script, but it looks better. Perhaps this could be the problem:

```
#94  0x420004ba in ppmEventCallBack(rmt_channel_t*, rmt_tx_done_event_data_t const*, void*) at C:\Users\Francis\AppData\Local\Programs\Arduino IDE/C:\Users\Francis\Documents\Arduino\libraries\ESP32_ppm-main\src/ESP32_ppm_TX.cpp:165
```

<details><summary>Click to see the full output</summary>
<p>

```
Found 124 unique addresses.
#0  0x403813ea in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:468
#1  0x403872b2 in __ubsan_include at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/ubsan.c:311
#2  0x3fc93af0 in ?? at ??:0
#3  0x3fc8fc00 in __global_pointer$ at ??:?
#4  0x3fca7500 in ?? at ??:0
#5  0x37363534 in ?? at ??:0
#6  0x7271706f in ?? at ??:0
#7  0x33323130 in ?? at ??:0
#8  0x00000001 in ?? at ??:0
#9  0x3fc93c9e in ?? at ??:0
#10  0x3fc93b54 in ?? at ??:0
#11  0x3fc92011 in __global_pointer$ at ??:?
#12  0x00000029 in ?? at ??:0
#13  0x3fc98000 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:88
#14  0x7a797877 in ?? at ??:0
#15  0x76757473 in ?? at ??:0
#16  0x00000049 in ?? at ??:0
#17  0x3fc93b48 in ?? at ??:0
#18  0x3fce0000 in ?? at ??:0
#19  0x3ff1b000 in ?? at ??:0
#20  0x3fcdf934 in ?? at ??:0
#21  0x00000000 in ?? at ??:0
#22  0x6e6d6c6b in ?? at ??:0
#23  0x6a696867 in ?? at ??:0
#24  0x66656463 in ?? at ??:0
#25  0x62613938 in ?? at ??:0
#26  0x00001881 in ?? at ??:0
#27  0x40380001 in _vector_table at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/riscv/vectors_intc.S:54
#28  0x00000002 in ?? at ??:0
#29  0x3c0d60fc in ?? at ??:0
#30  0x4038dcd6 in newlib_include_assert_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:92
#31  0x3fc92010 in __global_pointer$ at ??:?
#32  0x30343531 in ?? at ??:0
#33  0x3fc92378 in __global_pointer$ at ??:?
#34  0x3fc927c0 in __global_pointer$ at ??:?
#35  0x3c0c87b6 in ?? at ??:0
#36  0x3fc92388 in __global_pointer$ at ??:?
#37  0x3fc93b1c in ?? at ??:0
#38  0x3fc9238c in __global_pointer$ at ??:?
#39  0x3c0c895c in ?? at ??:0
#40  0x65737361 in ?? at ??:0
#41  0x66207472 in ?? at ??:0
#42  0x656c6961 in ?? at ??:0
#43  0x78203a64 in ?? at ??:0
#44  0x75657551 in ?? at ??:0
#45  0x63655265 in ?? at ??:0
#46  0x65766965 in ?? at ??:0
[... 58 lines cut ...]
#105  0x40385c14 in systimer_ticks_to_us at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32c3/systimer.c:16
#106  0x40385c20 in systimer_us_to_ticks at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32c3/systimer.c:21
#107  0x3fc93d68 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c:1041
#108  0x3fc93e08 in ?? at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c:157
#109  0x3fc93df8 in ?? at ??:0
#110  0x3fc93d24 in ?? at ??:0
#111  0x3fc93d38 in ?? at ??:0
#112  0x3fc9e948 in ?? at ??:0
#113  0x0000000a in ?? at ??:0
#114  0x00000010 in ?? at ??:0
#115  0x0001ffff in ?? at ??:0
#116  0x3fc93e10 in ?? at ??:0
#117  0x3fc93e24 in ?? at ??:0
#118  0x3fc93e30 in ?? at ??:0
#119  0x3fc93e48 in ?? at ??:0
#120  0x3fc93e5c in ?? at ??:0
#121  0x3fc93e8c in ?? at ??:0
#122  0x3fc93ea4 in ?? at ??:0
#123  0x3fc93eb8 in ?? at ??:0
```

</p>
</details>

I am going to keep an eye on this problem, but I cannot promise a quick fix for it.

fanfanlatulipe26, 2025-04-30:
Thank you for these results. I'll have a look.   
 The line that you highlighted is in a new library I developed recently using the RMT peripheral of ESP and the ESP-NOW  Espressif protocol.    It is the very last line ( only a {  ) following a return in a function callback.
 Maybe I made some touchy programming...  
My problem is that the application that I am developping based on this library is running pretty well often during minutes/hours and suddenly crash ...  I don't see the crash on ESP32. Only on the C3.    
From the application I derived a dummy test case that seems to crash more often: it is the one I used to make the zip file.

kittaakos, 2025-05-01:
> The line that you highlighted is in a new library I developed 

Thanks for using the decoder for your lib development

I have been tweaking the logic a bit further using gdb only, and the decoder can produce this fallback trace by getting the info for every distinct address:

```
#  0 0x403813ea in panic_abort+18 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:468
#  1 0x403872b2 in __ubsan_include at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/ubsan.c:311
#  2 xIsrStack+1568 in ?? ()
#  3 coex_schm_ble_mesh_config_bt_default_wifi_conn+8 in ?? ()
#  4 0x3fca7500 in ?? ()
#  5 0x37363534 in ?? ()
#  6 0x7271706f in ?? ()
#  7 0x33323130 in ?? ()
#  8 0x00000001 in ?? ()
#  9 xIsrStack+1998 in ?? ()
# 10 xIsrStack+1668 in ?? ()
# 11 0x3fc92011 in ?? ()
# 12 0x00000029 in ?? ()
# 13 s_timer_task in ?? ()
# 14 0x7a797877 in ?? ()
# 15 0x76757473 in ?? ()
# 16 0x00000049 in ?? ()
# 17 xIsrStack+1656 in ?? ()
# 18 0x3fce0000 in ?? ()
# 19 0x3ff1b000 in ?? ()
# 20 0x3fcdf934 in g_intr_lock_mux in ROM
# 21 0x00000000 in ?? ()
# 22 0x6e6d6c6b in ?? ()
# 23 0x6a696867 in ?? ()
# 24 0x66656463 in ?? ()
# 25 0x62613938 in ?? ()
# 26 0x00001881 in ?? ()
# 27 0x40380000 in _vector_table at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/riscv/vectors_intc.S:54
# 28 0x00000002 in ?? ()
# 29 __func__.10 in ?? ()
# 30 0x4038dcd6 in newlib_include_assert_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:92
# 31 0x3fc92010 in ?? ()
# 32 0x30343531 in ?? ()
# 33 0x3fc92378 in ?? ()
# 34 0x3fc927c0 in ?? ()
# 35 0x3c0c87b6 in ?? ()
# 36 0x3fc92388 in ?? ()
# 37 xIsrStack+1612 in ?? ()
# 38 0x3fc9238c in ?? ()
# 39 0x3c0c895c in ?? ()
# 40 0x65737361 in ?? ()
# 41 0x66207472 in ?? ()
# 42 0x656c6961 in ?? ()
# 43 0x78203a64 in ?? ()
# 44 0x75657551 in ?? ()
# 45 0x63655265 in ?? ()
# 46 0x65766965 in ?? ()
# 47 0x65757120 in ?? ()
# 48 0x632e6575 in ?? ()
# 49 0x3435313a in ?? ()
# 50 0x21282030 in ?? ()
# 51 0x20282028 in ?? ()
# 52 0x73615478 in ?? ()
# 53 0x7465476b in ?? ()
# 54 0x65686353 in ?? ()
# 55 0x656c7564 in ?? ()
# 56 0x61745372 in ?? ()
# 57 0x29286574 in ?? ()
# 58 0x203d3d20 in ?? ()
# 59 0x65736142 in ?? ()
# 60 0x65707954 in ?? ()
# 61 0x2920745f in ?? ()
# 62 0x29203020 in ?? ()
[... 39 lines cut ...]
#102 0x3fca745c in ?? ()
#103 0x40380270 in _interrupt_handler+148 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/riscv/vectors.S:369
#104 0x60023000 in ?? ()
#105 0x40385c14 in systimer_ticks_to_us at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32c3/systimer.c:16
#106 0x40385c20 in systimer_us_to_ticks at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/port/esp32c3/systimer.c:21
#107 ucStaticTimerQueueStorage.15 in ?? ()
#108 xActiveTimerList2 in ?? ()
#109 ucStaticTimerQueueStorage.15+144 in ?? ()
#110 xStaticTimerQueue.14+24 in ?? ()
#111 xStaticTimerQueue.14+44 in ?? ()
#112 0x3fc9e948 in ?? ()
#113 0x0000000a in ?? ()
#114 0x00000010 in ?? ()
#115 0x0001ffff in ?? ()
#116 xActiveTimerList2+8 in ?? ()
#117 xActiveTimerList1+8 in ?? ()
#118 s_common_recursive_mutex in ?? ()
#119 s_common_recursive_mutex+24 in ?? ()
#120 s_common_recursive_mutex+44 in ?? ()
#121 s_common_mutex in ?? ()
#122 s_common_mutex+24 in ?? ()
#123 s_common_mutex+44 in ?? ()
```

The updated logic retrieves info from https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/components/esp_rom/esp32c3/ld/esp32c3.rom.ld. I am not an ESP32 dev. Do you consider the updated version more helpful, or is it just more decoded lines that do not help anything? I want to understand if adding such a feature would eventually help the development or just add unnecessary code to this project.

fanfanlatulipe26, 2025-05-02:
Don’t know what to answer …    
Does it mean that you try to match the values on the stack with the addresses of the symbols you retrieve from the .ld files  ? (and so the address of this symbols are fixed for a given esp-idf release ?)
This information may be interesting for people working deep in the core , but difficult to interpret for a developer such as I am.

kittaakos, 2025-05-02:
> Does it mean that you try to match the values on the stack with the addresses of the symbols you retrieve from the .ld files ? (and so the address of this symbols are fixed for a given esp-idf release ?)

Yes, that's part of it — but it’s not limited to just that. Every address gets queried through GDB for details like address, line, symbol, functions, variables, and even source list if possible. For addresses that don't match the application's source code, the logic falls back to a ROM symbol list, similar to how ESP-IDF handles it:
https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/docs/en/api-guides/tools/idf-monitor.rst?plain=1#L185

> and so the address of this symbols are fixed for a given esp-idf release ?

I haven't figured it all out fully yet, but yes, in theory, a static lookup table per target and release could be maintained and used as a last-resort fallback.

> Don’t know what to answer …
> 
> This information may be interesting for people working deep in the core , but difficult to interpret for a developer such as I am.

Thanks for the honest feedback — I appreciate it. I'll leave the issue open in case it proves helpful later.

By the way, I'm working on a side project to decode ESP coredumps — crash happens, dump + ELF get sent to a server and decoded. Still rough, but if you ever hit a crash and can share the dump + ELF, I'd love to see if we can extract more from it:
https://github.com/dankeboy36/trbr/issues/19