dankeboy36/esp-exception-decoder #43: it reports `Could not recognize stack trace/backtrace` C6, RISC-V

kittaakos, 2025-05-09:
> Thank @kittaakos . I've already read the guide, and could do it well with xtensa core. But with C6, RISC-V, it reports `Could not recognize stack trace/backtrace`
> 
> ```
> Core  0 register dump:
> MEPC    : 0x42006c3a  RA      : 0x420016ac  SP      : 0x4081b9a0  GP      : 0x4080fed4  
> TP      : 0x4081bae0  T0      : 0x40022494  T1      : 0x4080916c  T2      : 0x00000000  
> S0/FP   : 0x40812000  S1      : 0x40815000  A0      : 0x00000000  A1      : 0x00000000  
> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000  
> A6      : 0x600b1400  A7      : 0x00000000  S2      : 0x50003fec  S3      : 0x40818b58  
> S4      : 0x40815000  S5      : 0x00000010  S6      : 0x00000000  S7      : 0x00000000  
> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
> T3      : 0x00000014  T4      : 0x40815000  T5      : 0x40815000  T6      : 0x00000000  
> MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000008  
> MHARTID : 0x00000000  
> 
> Stack memory:
> 4081b9a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
> 4081b9c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
> 4081b9e0: 0xffffffff 0x00000010 0xffffffff 0x00000000 0x00010000 0x00000000 0xffffffff 0x4080f788
> 4081ba00: 0x00000001 0x00000010 0xffffffff 0xffffffff 0x00000000 0x4080f788 0x00000000 0x4200ec1c
> 4081ba20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x00000011 0x00000010
> 4081ba40: 0xffffffff 0xffffffff 0x00000000 0x00000000 0x4080f788 0x00000000 0x00000001 0x4200ef98
> 4081ba60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xba7c61b8
> 4081ba80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081baa0: 0x00000000 0x00000000 0x00000000 0x4200fb90 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bac0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
> 4081bae0: 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x0000016c 0xabba1234 0x00000160 0x4081b7b0 0x00000000
> 4081bb00: 0x40811f84 0x40811f84 0x4081baf8 0x40811f7c 0x00000018 0x4081c448 0x4081c448 0x4081baf8
> 4081bb20: 0x00000000 0x00000001 0x40819ae8 0x706f6f6c 0x6b736154 0x00000000 0x4081bae0 0x00000005
> 4081bb40: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000571
> 4081bb60: 0x00000000 0x40815824 0x4081588c 0x408158f4 0x00000000 0x00000000 0x00000001 0x00000000
> 4081bb80: 0x00000000 0x00000000 0x4201977a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bba0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bbc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bbe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bc00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bc20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bc40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xbaad5678 0x00000068
> 4081bc60: 0xabba1234 0x0000005c 0x00000000 0x4081bc68 0x00000000 0x00000000 0x00000000 0x4081bc80
> 4081bc80: 0xffffffff 0x4081bc80 0x4081bc80 0x00000000 0x4081bc94 0xffffffff 0x4081bc94 0x4081bc94
> 4081bca0: 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000 0x00000000 0x00000001 0x00000000
> 4081bcc0: 0x00000000 0xbaad5678 0x00000088 0xabba1234 0x0000007c 0x00000000 0x00000014 0x4081c488
> 4081bce0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4081bd60 0x00000000 0x00000101
> 4081bd00: 0x00000000 0x00000000 0x0000000a 0x4081c52c 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bd20: 0x00000000 0x00000000 0x00000000 0x00000000 0x4081bf70 0x4081c0dc 0x00000000 0x4081c2d8
> 4081bd40: 0x4081c26c 0x4081c41c 0x4081c3b0 0x4081c344 0xbaad5678 0x0000020c 0xabba1234 0x00000200
> 4081bd60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 4081bd80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
> 
> Could not recognize stack trace/backtrace
> 
> Paste exception to decode...
> ``` 

 _Originally posted by @bhcuong2008 in [#6](https://github.com/dankeboy36/esp-exception-decoder/issues/6#issuecomment-2865436290)_

Please share the ELF file by attaching it as a zip (or any other allowed extension) if you can. Enable as much logging as possible (https://github.com/dankeboy36/esp-exception-decoder/issues/10#issuecomment-1727811557), and share the details. Please also copy-paste the entire content of the decoder terminal. If it's shareable info, I would like to see it with sketch path, FQBN, etc.

The more you can share about your environment, the more likely I will be able to fix the problem. If only the monitor output is available, I can check if the parser part of the code is correct, but I won't be able to debug the gdb part 😕  Thanks!

bhcuong2008, 2025-05-09:
Here is app built, flashed at 0x30000, that decoder reports `Could not recognize stack trace/backtrace`

[app_0x30000.zip](https://github.com/user-attachments/files/20119366/app_0x30000.zip)

bhcuong2008, 2025-05-09:
Here is app built, flashed at 0x10000, using default partition scheme of ESP32C6 Dev, that could show some of decode, but not having line detail.

```
Exception 7: Store/AMO access fault
MEPC: 0x420000a2
MTVAL: 0x00000000

[app_0x10000.zip](https://github.com/user-attachments/files/20119494/app_0x10000.zip)

Decoding stack results
0x00000000: ??
```

bhcuong2008, 2025-05-09:
Here is the source file .ino, 

class Person {
public:
    Person() {
        Serial.printf("New person\n");
    }
    void hello(const char* msg) {
        Serial.printf("%s\n", msg);
    }
};

Person *p = NULL;
Person *p2 = NULL;
int *p3 = NULL;

void setup() {   
    Serial.begin(115200);
    Serial.printf("Test exception, p = 0x%X | p2 = 0x%X\n", p, p2);
    p->hello("Hello, World 1!");    // not cause exception, WHY?
    p2->hello("Hello, World 2!");   // not cause exception, WHY?
    *p3 = 10;   // Cause exception here
}

void loop() {    
    delay(2000);
}

[WireScan.zip](https://github.com/user-attachments/files/20119678/WireScan.zip)

bhcuong2008, 2025-05-09:
I install Arduino ESP32 core via command line, using version 3.2.0, in Ubuntu environment.

bhcuong2008, 2025-05-09:
To build app with flash address different from 0x10000, it must change manually file hardware/espressif/esp32/platform.txt to update  0x10000 with new value 0x30000.

kittaakos, 2025-05-12:
I confirm that compiling, uploading, monitoring, and decoding the exception provides insufficient traces:

```
ESP Exception Decoder
Sketch: eed_issue43 FQBN: esp32:esp32:esp32c6

Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420000a2  RA      : 0x42002024  SP      : 0x408161d0  GP      : 0x4080bdd4  
TP      : 0x40816220  T0      : 0x400283c2  T1      : 0x40806910  T2      : 0x00000000  
S0/FP   : 0x00000000  S1      : 0x42027000  A0      : 0x00000010  A1      : 0x00000000  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x0000000a  A5      : 0x00000000  
A6      : 0x600b1400  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x4081699c  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
408161d0: 0x00000000 0x00000000 0x00000000 0x42002024 0x00000000 0x00000000 0x00000000 0x40806c52
408161f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816210: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678
40816230: 0x00000168 0xabba1234 0x0000015c 0x40815fc0 0x00000000 0x4080d534 0x4080d534 0x4081623c
40816250: 0x4080d52c 0x00000018 0x00000000 0x00000000 0x4081623c 0x00000000 0x00000001 0x4081422c
40816270: 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x40816220 0x00000004 0x00000000 0x00000001
40816290: 0x00000000 0x00000000 0x00000000 0x00000106 0x00000000 0x4080f1c4 0x4080f22c 0x4080f294
408162b0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4200b42c 0x00000000
408162d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408162f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816310: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816330: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816350: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816370: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816390: 0x00000000 0x00000000 0xbaad5678 0x00000068 0xabba1234 0x0000005c 0x00000000 0x408163a8
408163b0: 0x00000000 0x00000000 0x00000000 0x408163c0 0xffffffff 0x408163c0 0x408163c0 0x00000000
408163d0: 0x408163d4 0xffffffff 0x408163d4 0x408163d4 0x00000001 0x00000001 0x00000000 0xa500ffff
408163f0: 0x00000000 0xa5a5a5a5 0xa5a5a501 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000088 0xabba1234
40816410: 0x0000007c 0x00000000 0x00000014 0x40816c6c 0x00000000 0x00000000 0x00000000 0x00000000
40816430: 0x00000000 0x408164a0 0x00000000 0x00000101 0x00000000 0x00000000 0x0000000a 0x40816bc8
40816450: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816470: 0x408166b0 0x4081681c 0x00000000 0x40816a18 0x408169ac 0x40816b5c 0x40816af0 0x40816a84
40816490: 0xbaad5678 0x0000020c 0xabba1234 0x00000200 0x00000000 0x00000000 0x00000000 0x00000000
408164b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408164d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408164f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816530: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816550: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816570: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40816590: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
408165b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 3206192b6

Exception 7: Store/AMO access fault
MEPC: 0x420000a2
MTVAL: 0x00000000

Decoding stack results
0x526c8040: ??

Paste exception to decode...
```

> Here is app built, flashed at 0x30000

> Here is app built, flashed at 0x10000

I can't comment on these, unfortunately.

There is another project (https://github.com/dankeboy36/trbr) that provides slightly better decoded output:

```
Core 0 | Store/AMO access fault | 7

PC → 0x420000a2: setup() at /Users/kittaakos/dev/sandbox/trbr/.tests/sketches/eed_issue43/eed_issue43.ino:20
Addr → 0x00000000

0x526c8040: ??Waiting for the debugger to disconnect...
```

When looking at the stdout from gdb, it looks like this:

```
0x420000a2 in setup () at /Users/kittaakos/dev/sandbox/trbr/.tests/sketches/eed_issue43/eed_issue43.ino:20
20	  *p3 = 10;                      // Cause exception here
#0  0x420000a2 in setup () at /Users/kittaakos/dev/sandbox/trbr/.tests/sketches/eed_issue43/eed_issue43.ino:20
#1  0x42002024 in loopTask (pvParameters=<optimized out>) at /Users/kittaakos/Library/Arduino15/packages/esp32/hardware/esp32/3.2.0/cores/esp32/main.cpp:59
#2  0x526c8040 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

I will try to get the improvements from trbr into this project (https://github.com/dankeboy36/esp-exception-decoder/pull/40), please bear with me.

bhcuong2008, 2025-05-13:
@kittaakos thank you for ur effort. It's a good news that your trbr project has better output.
I'm looking forward to your improvements from trbr💯

dankeboy36, 2025-07-06:
> I'm looking forward to your improvements from trbr💯

A tester VSIX with a complete rewrite of the decoder logic is available at https://github.com/dankeboy36/esp-exception-decoder/pull/40#issue-2958896427.

bhcuong2008, 2025-07-07:
@dankeboy36 , I've just tested file at https://github.com/dankeboy36/esp-exception-decoder/raw/refs/heads/feat/use-trbr/esp-exception-decoder-1.1.0.vsix, but still can't decode it, with the same result above.
```
Exception 7: Store/AMO access fault
MEPC: 0x420000a2
MTVAL: 0x00000000

Decoding stack results
0x00000000: ??
```

dankeboy36, 2025-07-07:
> but still can't decode it, with the same result above.

Thank you for your prompt help.

From the decoder output, I can see that Arduino IDE 2.x didn’t pick up the new version. It currently shows `Exception 7: Store/AMO access fault` instead of the expected `0 | Store/AMO access fault | 7`.

However, I’ve noticed a GDB tool path discovery issue on Windows. Let me address that first, and then I’ll assist you with installing the correct decoder version.

Thanks for your patience

bhcuong2008, 2025-07-08:
@dankeboy36 I've tried many tests, but I could not figure out why.
I download this file, and put it in plugin folder.
https://github.com/dankeboy36/esp-exception-decoder/raw/refs/heads/feat/use-trbr/esp-exception-decoder-1.1.1-preview.1.vsix

I tested 2 cases, with Arduino IDE 2.3.6, Arduino ESP32 3.2.1, ESP-IDF 5.4.2 using partition scheme "Default 4MB with spiffs"
**1. With board "ESP32C3 Dev Module", FQBN esp_cth:esp32:esp32c3, original from Board select**
It could decode well with right result
```
0 | Store/AMO access fault | 7

PC -> 0x420000a2: setup () at /cfm/device/c200_esp32c3/sketch/WireScan/WireScan.ino:21

0x420000a2: setup () at /cfm/device/c200_esp32c3/sketch/WireScan/WireScan.ino:21
0x42001fc6: loopTask (pvParameters=<optimized out>) at /cfm/device/c200_esp32c3/sketch/hardware/esp_cth/esp32/cores/esp32/main.cpp:59
0x00000000: ??
```

2. With board copied from "ESP32C3 Dev Module", FQBN esp_cth:esp32:esp32c3_test, without any changes vs the original one.
It could not decode. It reports only one line, `No register addresses found to decode`

Could you give me some hints for this case. Does it depends on FQBN in decoding process?

kittaakos, 2025-07-08:
> **2. With board copied from "ESP32C3 Dev Module", FQBN esp_cth:esp32:esp32c3_test, without any changes vs the original one** It could not decode. It reports only one line, `No register addresses [...]

Hello, I am very lucky that you're supporting me. The issue is with the FQBN, specifically with the last part, the board ID. The decoder determines that it needs to switch to RISC-V decoding mode based on the board ID. Currently, the supported board IDs for RISC-V decoding are: `esp32c2`, `esp32c3`, `esp32c6`, `esp32h2`, and `esp32h4`. Otherwise, it uses Xtensa (Details: https://github.com/dankeboy36/trbr?tab=readme-ov-file#decode-using-gdb). Please try adjusting the board ID in the cloned boards.txt (?) to one of these supported RISC-V board IDs and let me know if it works. I will consider relaxing the logic based on the board ID, attempting to guess if it cannot decode anything, and falling back to the other architecture type.

bhcuong2008, 2025-07-08:
Could you support board ID with pattern esp32c3_a_b_c. You split board ID by "_", and get the 1st part to decide which kind of chips. For example, to my test, ID is esp32c3_test, now split it by "_" to get the 1st part esp32c3.

Because I/We must reserve the original one from Arduino ESP32 board definitions for testing easily.

bhcuong2008, 2025-07-08:
For my own testing and development, I copy folder espressif to esp_cth, so that when I need to test on the original from Arduino ESP32, I will switch to it in Arduino IDE. This does not affect anything. 

About board ID, I often copy the standard definition of Arduino ESP32 such as ESP32C3 Dev Module, ESP32C6 Dev Module to my own in the same file boards.txt due to I use different partition schemes, and other options. I often name custom board ID like that.

As you see in the board definitions, there are many custom boards, such as "TTGO T-OI PLUS RISC-V ESP32-C3", its board ID is `ttgo-t-oi-plus`. If someone chooses this board to develop, then he could not debug with this exception decoder.

So I think that it is common needs to have custom board definition.

bhcuong2008, 2025-07-08:
@kittaakos , I found that you could use this value, `build.mcu` to decide which chip to decode.

dankeboy36, 2025-07-08:
> board_id`.build.tarch`.

The ESP-IDF platform completely separates the RISC-V decoding logic for each C3, C6, H2, etc., although it remains the same. I want to keep the decoder extension close to it. While `build.mcu` will tell `esp32c3`, `board.tarch` will say only `riscv32`. I will add the changes before the end of this week. Thanks for supporting me with the testing

bhcuong2008, 2025-07-08:
@kittaakos Thank you so much

dankeboy36, 2025-07-10:
> I found that you could use this value, `build.mcu` to decide which chip to decode. Not depend on board ID.

Please try with https://github.com/dankeboy36/esp-exception-decoder/raw/refs/heads/feat/use-trbr/esp-exception-decoder-1.1.1-preview.2.vsix. Thank you

dankeboy36, 2025-07-10:
> I've already tested it. It's very good now.

You're the best. I appreciate your support.